realsenseai/librealsense #1525: Error building librealsense on Ubuntu 16.04

gregorsamsa183, 2018-04-11:
While trying to build on ubuntu I'm getting the error: error: ‘glfwSetDropCallback’ was not declared in this scope. 

Glfw is set in my usr/include/lib and usr/include folders and it is properly recognized in Cmake GUI as well. I'm using the option with Python and my current Python version is 3.5. 

The full trace of the error is: 

[ 91%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/model-views.cpp.o
/home/burcak/Desktop/librealsense/tools/rosbag-inspector/rs-rosbag-inspector.cpp: In member function ‘void gui_window::init_window()’:
/home/burcak/Desktop/librealsense/tools/rosbag-inspector/rs-rosbag-inspector.cpp:95:10: error: ‘glfwSetDropCallback’ was not declared in this scope
         });
          ^
tools/rosbag-inspector/CMakeFiles/rs-rosbag-inspector.dir/build.make:62: recipe for target 'tools/rosbag-inspector/CMakeFiles/rs-rosbag-inspector.dir/rs-rosbag-inspector.cpp.o' failed
make[2]: *** [tools/rosbag-inspector/CMakeFiles/rs-rosbag-inspector.dir/rs-rosbag-inspector.cpp.o] Error 1
CMakeFiles/Makefile2:1287: recipe for target 'tools/rosbag-inspector/CMakeFiles/rs-rosbag-inspector.dir/all' failed
make[1]: *** [tools/rosbag-inspector/CMakeFiles/rs-rosbag-inspector.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 91%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/ux-window.cpp.o
[ 91%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/ux-window.cpp.o
/home/burcak/Desktop/librealsense/common/ux-window.cpp: In member function ‘void rs2::ux_window::open_window()’:
/home/burcak/Desktop/librealsense/common/ux-window.cpp:89:10: error: ‘glfwSetDropCallback’ was not declared in this scope
         });
          ^
/home/burcak/Desktop/librealsense/common/ux-window.cpp: In member function ‘void rs2::ux_window::open_window()’:
/home/burcak/Desktop/librealsense/common/ux-window.cpp:89:10: error: ‘glfwSetDropCallback’ was not declared in this scope
         });
          ^
[ 92%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/model-views.cpp.o
tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/build.make:182: recipe for target 'tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/ux-window.cpp.o' failed
make[2]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/ux-window.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 92%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/third-party/imgui/imgui.cpp.o
tools/depth-quality/CMakeFiles/rs-depth-quality.dir/build.make:110: recipe for target 'tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/ux-window.cpp.o' failed
make[2]: *** [tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/ux-window.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1121: recipe for target 'tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all' failed
make[1]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all] Error 2
CMakeFiles/Makefile2:1231: recipe for target 'tools/depth-quality/CMakeFiles/rs-depth-quality.dir/all' failed
make[1]: *** [tools/depth-quality/CMakeFiles/rs-depth-quality.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

dorodnic, 2018-04-11:
Hi @gregorsamsa183 
Most likely there is a mismatch in the version of installed GLFW. 
Did you run `sudo apt-get install libglfw3-dev` or `./scripts/install_glfw3.sh`?

gregorsamsa183, 2018-04-11:
I did sudo apt-get install libglfw3-dev because it was on your installation directions. 

Anyway, I removed the previous installation from the command line and did the steps in the install script. Now it's working. Thanks!

closed 2018-04-11 by gregorsamsa183