realsenseai/librealsense #1526: Misalignment between IR Y16 and Y8 images

panjekm, 2018-04-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  v2.10.1                        | 

### Issue Description
When capturing the Y16 and Y8 IR images in realsense-viewer I observed that these two images are not the same. They seems to have a small shift and scale change (see below). What are the transformations applied to the Y8 IR image?

[image]

[image]

And here is the diff between the above two images:
[image]

panjekm, 2018-04-16:
Any update as to what this shift/scaling is? There seems to be an even bigger shift between Y8 and Y16 1920x1080 IR images of the second IR camera.

This does not happen when comparing Y16 and Y8 images of lower resolutions.

panjekm, 2018-04-25:
Hi @JasonChang13 

No, its not the same issue. The one youre refering to was also opened by me, but it only had to do with bayer artifacts. There was no shifting or scaling involved when debayering.

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
hi @panjekm,

the Y16 image is unrectified image and Y8 image is rectified image, so they do different.

panjekm, 2018-05-30:
Can you double check this? 
When I look at both raw IR Y16 fullHD images they seem to be rectified already. The IR Y8 1280x720 images then seem to be just zoomed in. 

What transformations do you do between IR Y16 fullHD and IR Y8 1280x720 images? Also, when do you undistort them?

dorodnic, 2018-05-30:
Rectification is done by the ASIC and introduces a homography transformation from Y16 to Y8 view. 
It does not affect the distortion model of the frame, just reprojecting them into a shared virtual plane.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Is this still an issue?  If not, ticket will be closed in 5 days.

cartokk, 2018-06-22:
I am also interested to calibrate the cameras myself. 
1) Is there a way to receive the unrectified (i.e. raw) IR images? Is a simple change to Y16 in the SDK enough to receive "high quality" unrectified IR images? Is there any post-processing needed? 
2) What parameters (i.e. shift and zoom) does the ASIC (for D415/D435) use for rectification?
Thanks!

panjekm, 2018-06-24:
Its still unclear how exactly you do calibration. Im using openCV rectification function on IR1 and IR2 Y16 images I cant seem to get to the same resulting Y8 image. There is this shift/zoom difference between my rectified Y16 image and your rectified Y8 image.

cartokk, 2018-06-27:
@agrunnet 
Thanks a lot! Is it also possible to change the format with ros? I changed the format from Y8...

    _format[INFRA1] = RS2_FORMAT_Y8;   // libRS type
    _image_format[INFRA1] = CV_8UC1;    // CVBridge type
    _encoding[INFRA1] = sensor_msgs::image_encodings::TYPE_8UC1; // ROS message type
    _unit_step_size[INFRA1] = sizeof(uint8_t); // sensor_msgs::ImagePtr row step size
    _stream_name[INFRA1] = "infra1";
    _depth_aligned_encoding[INFRA1] = sensor_msgs::image_encodings::TYPE_16UC1;

...to Y16...

    _format[INFRA1] = RS2_FORMAT_Y16;   // libRS type
    _image_format[INFRA1] = CV_16UC1;    // CVBridge type
    _encoding[INFRA1] = sensor_msgs::image_encodings::TYPE_16UC1; // ROS message type
    _unit_step_size[INFRA1] = sizeof(uint16_t); // sensor_msgs::ImagePtr row step size
    _stream_name[INFRA1] = "infra1";
    _depth_aligned_encoding[INFRA1] = sensor_msgs::image_encodings::TYPE_16UC1;

However, the format Y16 doesn't seem to be supported. Any idea what I did wrong?

Thanks!

RealSense-Customer-Engineering, 2018-08-15:
[Realsense Customer Engineering Team Comment]
hi @panjekm,

i think you are comparing our Y8 and your Y8 image after rectified from our Y16 image, right?
maybe you can share how you do rectification and we can check if any difference.
also i think Y16 support only 1280x800, so suggest you to try again with this resolution.

dorodnic, 2018-10-24:
I think we can close this. Hopefully the differences between Y8 and Y16 were sufficiently illuminated. As for ROS, @cartokk feel free to open a separate issue if you need this. At the moment it is not supported I believe, just because we don't see any application for Y16 frames in a robot (with them being non-rectified and all)

closed 2018-10-24 by dorodnic