realsenseai/librealsense #1529: acceptable range of temperature for ASIC and Projector in D415

kyranf, 2018-04-12:
Camera Model: D415
OS: linux, Ubuntu 16.04.03 LTS with Kernel 4.4.102 Low Latency
Realsense SDK 2.9.1
Hello,

I have a system which has a computer and the D415 in a sealed box. It's able to fan-forced radiate enough heat to keep the computer CPU at 52 degrees Celsius , and the D415's ASIC temperature at 58 degrees Celsius , and the D415's projector at 55 degrees Celsius.

My computer is okay with up to 80 degrees operating temp, before issues start happening (it throttles due to the intel powerclamp drivers in Ubuntu). 

The ambient air inside I am guessing is around 35-40 degrees Celsius. 

What is the acceptable range of temperatures for the ASIC and projector as read from the realsense viewer program?

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
hi @kyranf,

please take D415 as single product and consider the temperature base on datasheet as below link:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf
at chapt. 3.3.7.

kyranf, 2018-04-17:
Okay, thanks. The ratings show 55 degrees ambient is the max operating temperature. that's enough for me I guess! The ASIC temperatures being ~60 degrees I guess is okay. Most silicon chips are fine up to 100-120 degrees or more anyway. I am worried about life expectancy  though :(

RealSense-Customer-Engineering, 2018-04-19:
[Realsense Customer Engineering Team Comment] 
Please refer to table 7-10 for Vision Processor Temp Limits (Based on Still Air) and its max is up to 110 degrees.
For the 55 degree ambient, that means the camera can operate at the ambient temperature 55 degrees.

kyranf, 2018-04-19:
Thank you, that answers all my questions.

closed 2018-04-19 by kyranf