realsenseai/librealsense #1530: How to get y coordinate

linf233, 2018-04-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Windows10| 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC|
| SDK Version                            |   2.10.2                       | 

### Issue Description
I try to read the y coordinate and generate a graph likes depthmap about the y coordinate, but I can't find the relevant information. Most of the information is about the z coordinate.

dorodnic, 2018-04-12:
Hi @linf233 
Please take a look at the following modified version of [rs-capture.cpp](https://github.com/dorodnic/librealsense/blob/text_sample/examples/capture/rs-capture.cpp).
In it, I'm iterating over all pixels in the depth image and calculate depth for every x, y.
Hope it can help.

linf233, 2018-04-12:
Thank you for your response @dorodnic 
But what I want is the Y coordinate in the real world instead of the X,Y coordinate of the pixel in the depthmap. I know that we can get the distance very quickly, that the distance in the z direction. So I'm trying to get "height map" about Y direction from depth map.

dorodnic, 2018-04-12:
Perhaps the [pointcloud](https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud) example can be useful? The output of `pointcloud` object is a collection of `{x,y,z}` coordinates in the real world. You can also calculate these yourself using SDK helper functions under [rsutil.h](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h)

linf233, 2018-04-12:
Thanks,I will try them.

dorodnic, 2018-04-16:
Hi @linf233 
I'm closing this issue for now. If you need further assistance don't hesitate to open a new one.

closed 2018-04-16 by dorodnic