realsenseai/librealsense #1532: Connectivity issues with depth stream

TheMikeyR, 2018-04-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04.4)  | 
| Kernel Version (Linux Only)    |  4.4.0-119-generic                                 | 
| Platform                                 | PC  |
| SDK Version                            |  v2.10.3                       | 
| Language                            |  C++                       | 

### Issue Description
I'm running the rs-capture example without any modifications. I've tested with two D415, one with original cable and one with original cable + 5m deltaco active usb3.0 extender. 

When I plug in the device I get 2-3 depth frames and then it chrashes, the rgb stream works fine and will continue but depth freeze. If I replug the cable without closing the program it will stream just fine on both devices. 

It seems to me, to get consistent depth stream frame I have to start program, let it freeze (takes 2-3 frames), then take out the USB and replug it. I have patched my drivers with your realsense script (I can get the hardware clock). I have the same issue on a non-supported kernel 4.14. 

It also seems if I get it to work, if I stop the script and start it again, it will hang after a couple of frames. The weird thing is that the RGB stream works fine. 

here is `udevadm monitor`:
```
KERNEL[1063.399955] remove   /2-6:1.3/media2 (media)
UDEV  [1063.400620] remove   /2-6:1.3/media2 (media)
KERNEL[1063.421059] remove   /2-6:1.0/input/input27/event12 (input)
UDEV  [1063.423801] remove   /2-6:1.0/input/input27/event12 (input)
KERNEL[1063.449770] remove   /2-6:1.0/input/input27 (input)
KERNEL[1063.450377] remove   /2-6:1.0/media1 (media)
UDEV  [1063.451816] remove   /2-6:1.0/input/input27 (input)
UDEV  [1063.451910] remove   /2-6:1.0/media1 (media)
KERNEL[1063.451976] remove   /2-6:1.3/media4 (media)
UDEV  [1063.452645] remove   /2-6:1.3/media4 (media)
KERNEL[1063.488923] remove   /2-6:1.0/input/input26/event19 (input)
UDEV  [1063.491394] remove   /2-6:1.0/input/input26/event19 (input)
KERNEL[1063.509821] remove   /2-6:1.0/input/input26 (input)
KERNEL[1063.510483] remove   /2-6:1.0/media3 (media)
UDEV  [1063.511586] remove   /2-6:1.0/input/input26 (input)
UDEV  [1063.512752] remove   /2-6:1.0/media3 (media)
KERNEL[1096.576612] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video7 (video4linux)
KERNEL[1096.576739] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video8 (video4linux)
UDEV  [1096.579390] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video7 (video4linux)
UDEV  [1096.581201] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video8 (video4linux)
KERNEL[1096.600575] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28/event20 (input)
UDEV  [1096.602924] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28/event20 (input)
KERNEL[1096.616312] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28 (input)
KERNEL[1096.616401] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/media5 (media)
KERNEL[1096.616482] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
KERNEL[1096.616559] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.1 (usb)
KERNEL[1096.616636] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.2 (usb)
KERNEL[1096.616796] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/video4linux/video9 (video4linux)
UDEV  [1096.617398] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28 (input)
UDEV  [1096.617431] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/media5 (media)
KERNEL[1096.617452] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/media6 (media)
KERNEL[1096.617477] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3 (usb)
KERNEL[1096.617519] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.4 (usb)
UDEV  [1096.617756] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.1 (usb)
UDEV  [1096.618127] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
UDEV  [1096.618182] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/media6 (media)
UDEV  [1096.618385] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.2 (usb)
UDEV  [1096.618411] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.4 (usb)
UDEV  [1096.618449] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/video4linux/video9 (video4linux)
KERNEL[1096.618753] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
UDEV  [1096.618947] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3 (usb)
UDEV  [1096.621857] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
KERNEL[1101.833491] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
KERNEL[1101.835229] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
KERNEL[1101.835545] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/media1 (media)
KERNEL[1101.837225] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video1 (video4linux)
KERNEL[1101.838876] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video2 (video4linux)
[... 101 lines cut ...]
KERNEL[1152.313392] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.4 (usb)
UDEV  [1152.341721] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2 (usb)
UDEV  [1152.345747] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.0 (usb)
UDEV  [1152.346744] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.1 (usb)
UDEV  [1152.347049] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.2 (usb)
UDEV  [1152.348324] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.0/media1 (media)
UDEV  [1152.348558] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.4 (usb)
UDEV  [1152.348622] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.3 (usb)
UDEV  [1152.349283] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.0/input/input31 (input)
UDEV  [1152.349707] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.3/media2 (media)
UDEV  [1152.350245] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.0/video4linux/video1 (video4linux)
UDEV  [1152.350318] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.0/video4linux/video2 (video4linux)
UDEV  [1152.352187] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.3/video4linux/video3 (video4linux)
UDEV  [1152.406131] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6.2/2-6.2:1.0/input/input31/event12 (input)
KERNEL[1259.712587] remove   /1-6:1.0/input/input30/event19 (input)
UDEV  [1259.713405] remove   /1-6:1.0/input/input30/event19 (input)
KERNEL[1259.724616] remove   /1-6:1.0/input/input30 (input)
KERNEL[1259.724764] remove   /1-6:1.0/media3 (media)
UDEV  [1259.725224] remove   /1-6:1.0/input/input30 (input)
UDEV  [1259.725307] remove   /1-6:1.0/media3 (media)
KERNEL[1264.007885] add      /kernel/slab/nf_conntrack_4 (slab)
UDEV  [1264.009504] add      /kernel/slab/nf_conntrack_4 (slab)
KERNEL[1264.040597] add      /kernel/slab/nf_conntrack_5 (slab)
UDEV  [1264.041204] add      /kernel/slab/nf_conntrack_5 (slab)
```

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
From the log you provide, it seems the device is detected as either USB2 device rather than USB3 device that provide sufficient bandwidth.
it could the USB signal quality is not good enough that the PC (host) enumerate the device as  USB2 device.
you can check if the device is enumerate as USB3 device when issue is not there
if that is the case you can change cable or use shorter cable to see if it can help.

TheMikeyR, 2018-04-18:
I've tried several of the included cables, and my own, and an active extender all same results. 

Works fine in intel realsense it's only in the examples, where e.g. rs-capture, it gets two first frames, freeze, then crash because no frame within 5 seconds. If I replug the cable before it crash, it restores itself and shows video feed without any problems. For some reason I need to plug it in, start program, wait for freeze, plug it out and in again. Really annoying to develop with.

Also tried with kernel 4.14.13-041413-generic without any changes. 

I've tried all my ports in both laptop (lenovo thinkpad p51) and the dock, all ports are usb3. Also tried an usb3 hub, which I use for jetosn. Still no changes in the process. 

Here is add, remove to replug udevadm monitor

```
KERNEL[1700.393687] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
KERNEL[1700.395420] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
KERNEL[1700.397462] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video1 (video4linux)
KERNEL[1700.399125] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video2 (video4linux)
KERNEL[1700.399213] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/media13 (media)
KERNEL[1700.399418] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28 (input)
KERNEL[1700.399497] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28/event27 (input)
KERNEL[1700.399728] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.1 (usb)
KERNEL[1700.399915] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.2 (usb)
KERNEL[1700.400059] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3 (usb)
KERNEL[1700.401929] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/video4linux/video3 (video4linux)
KERNEL[1700.401993] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/media14 (media)
KERNEL[1700.402230] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.4 (usb)
UDEV  [1700.428138] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
UDEV  [1700.428630] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
UDEV  [1700.428915] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.1 (usb)
UDEV  [1700.429090] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.2 (usb)
UDEV  [1700.430322] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3 (usb)
UDEV  [1700.430467] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.4 (usb)
UDEV  [1700.430767] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/media13 (media)
UDEV  [1700.430911] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28 (input)
UDEV  [1700.431474] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/media14 (media)
UDEV  [1700.431696] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video1 (video4linux)
UDEV  [1700.431927] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video2 (video4linux)
UDEV  [1700.432858] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/video4linux/video3 (video4linux)
UDEV  [1700.527695] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input28/event27 (input)
KERNEL[1711.304680] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video1 (video4linux)
KERNEL[1711.304764] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video2 (video4linux)
KERNEL[1711.304841] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
KERNEL[1711.305079] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.1 (usb)
KERNEL[1711.305178] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.2 (usb)
KERNEL[1711.305253] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/video4linux/video3 (video4linux)
KERNEL[1711.305318] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3 (usb)
KERNEL[1711.305478] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.4 (usb)
KERNEL[1711.306753] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
UDEV  [1711.307442] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video1 (video4linux)
UDEV  [1711.308846] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video2 (video4linux)
UDEV  [1711.309175] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.1 (usb)
UDEV  [1711.309432] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.2 (usb)
UDEV  [1711.310150] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.4 (usb)
UDEV  [1711.310192] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3/video4linux/video3 (video4linux)
UDEV  [1711.310217] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
UDEV  [1711.310239] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.3 (usb)
UDEV  [1711.312719] remove   /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
KERNEL[1712.865281] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6 (usb)
KERNEL[1712.866760] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0 (usb)
KERNEL[1712.868656] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video4 (video4linux)
KERNEL[1712.870255] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video5 (video4linux)
KERNEL[1712.870316] add      /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/media15 (media)
[... cut ...]
UDEV  [2028.491825] add      /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.4/2-4.4.3/2-4.4.3:1.0/video4linux/video2 (video4linux)
UDEV  [2028.491999] add      /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.4/2-4.4.3/2-4.4.3:1.3/video4linux/video3 (video4linux)
UDEV  [2028.492292] add      /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.4/2-4.4.3/2-4.4.3:1.0/video4linux/video1 (video4linux)
UDEV  [2028.542865] add      /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.4/2-4.4.3/2-4.4.3:1.0/input/input33/event256 (input)
```

TheMikeyR, 2018-04-18:
More debug info
```
 ~  sudo lsusb -vvv |grep -i -B5 -A5 bcdUSB

Bus 002 Device 004: ID 8086:0ad3 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
FIXME: alloc bigger buffer for device capability descriptors
--

Bus 002 Device 003: ID 17ef:305b Lenovo 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x17ef Lenovo
--

Bus 002 Device 002: ID 17ef:305a Lenovo 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x17ef Lenovo
--

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
--
```

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
@TheMikeyR
please see if the similar ticket can help on your issue
https://github.com/IntelRealSense/librealsense/issues/1391
if Dxxx is enumerated as USB2 device (even the PC port is USB3), bandwidth is smaller than USB3 that will cause streaming issue
you can use lsusb- t to check the enumerated device speed
USB1/2/3   12M/480M/5G

--Quote from  #1391#
There is a hardware issue with the type-C connector on the camera module that may be causing your problem. Can you please confirm that the issue still occurs if you reverse the orientation of the plug. Also try plugging the connector in quickly and/or plugging in the camera side and then the host side last. There is a controller on the board that switches a MUX to route the SS signals from the connector to the ASIC and it is slow to switch; the USB2 signals always connect first and if the ASIC doesn't see the USB3 signals soon enough, it starts in USB2 mode.
Let me know if this makes a difference. There is a hardware revision that should be released soon.

TheMikeyR, 2018-04-18:
Seems to be registered in USB3 mode according to `lsusb -t` it says 5000M when I connect the camera.

I've tried to reverse the orientation of the USB-C connector without any success, still same issue. (RGB stream works fine, but depth stream hangs until I replug while the program is still running). Intel realsense viewer works just fine, activating and disabling stream on the run. 

Without camera 

```
 ~  lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/3p, 12M
                |__ Port 3: Dev 11, If 2, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 3: Dev 11, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 3: Dev 11, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 3: Dev 11, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 2: Dev 9, If 1, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 2: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 8, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 8, If 2, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 8: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 8: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 9: Dev 5, If 0, Class=Vendor Specific Class, Driver=, 12M
    |__ Port 11: Dev 7, If 0, Class=Chip/SmartCard, Driver=, 12M
    |__ Port 13: Dev 10, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 14: Dev 12, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 12, If 0, Class=Wireless, Driver=btusb, 12M
```

with camera
```
 ~  lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
    |__ Port 6: Dev 4, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 6: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 6: Dev 4, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 6: Dev 4, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 6: Dev 4, If 4, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/3p, 12M
                |__ Port 3: Dev 11, If 2, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 3: Dev 11, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 3: Dev 11, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 3: Dev 11, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 2: Dev 9, If 1, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 2: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 8, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 8, If 2, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 8: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 8: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 9: Dev 5, If 0, Class=Vendor Specific Class, Driver=, 12M
    |__ Port 11: Dev 7, If 0, Class=Chip/SmartCard, Driver=, 12M
    |__ Port 13: Dev 10, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 14: Dev 12, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 12, If 0, Class=Wireless, Driver=btusb, 12M

```

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
Hi @TheMikeyR
Could you try the same rs-capture app on another PC with same Ubuntu config?
We have newer FW (v 5.9.2) which provide more robust streaming capability
https://downloadcenter.intel.com/download/27522/Latest-Firmware-for-Intel-RealSense-D400-Product-Family,  you can see if it can help on your case.
by the way, the issue only happen with rs-capture but not realsense-viewer, correct and only D415 is plugged, correct?

TheMikeyR, 2018-04-18:
The issue happens with all examples, but only c++ example, the python wrapper example works just fine. It's really odd, I've also tried the development branch without any luck. 

It doesn't happened with realsense-viewer and it is only d415 plugged in that port (I've also have usb mouse and keyboard plugged in another port). 

I would like to get it to work with c++ since python wrapper are missing stuff such as filters. 

I can test it on another ubuntu 16.04 on Friday. Meanwhile I will give the jetson tx2 a shot.

TheMikeyR, 2018-04-18:
It works fine with the examples on jetson tx2, so I guess it has something to do with my laptop. 
I've changed several kernels without any change in result, do you know what could potential be wrong ?

RealSense-Customer-Engineering, 2018-05-14:
[Realsense Customer Engineering Team Comment]
@TheMikeyR
had you tried with another desktop?
by the way, we have new Firmware that provide more robust streaming
please check below link about how to upgrade the FW
https://downloadcenter.intel.com/download/27408/?v=t

TheMikeyR, 2018-05-14:
I've upgraded firmware to Version: 5.9.9.2, and the issue still persist only with the c++, when using python it works just fine. 

gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.9)

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Is this still an issue?  If not, ticket will be closed in 5 days.

TheMikeyR, 2018-06-22:
@RealSense-Customer-Engineering thanks for reminding me of the issue, I've been using a different platform. 
I've just made the test again and it is no longer an issue on my laptop running. I have no clue how the issue have been solved since I didn't update the SDK for a while  (I had 2.11.1), and have also tested it on camera with old FW and new FW.

So I can happily say my issue has been fixed.

closed 2018-06-22 by TheMikeyR