realsenseai/librealsense #1534: Starting the camera blocks on Odroid with Android

kevindehecker, 2018-04-12:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Android 7.1.1  | 
| Kernel Version (Linux Only)    |  3.10.9-ga7accf5-dirty                                         | 
| Platform                                 | Odroid XU4  |
| SDK Version                            |  2.10.3                           | 

### Issue Description

I've pulled the latest development branch ( 7abab50 ), and when I attempt to run the rs-color example, it just gets stuck...:
 
```
odroidxu3:/data/lrs_binaries # ./rs-color                                      
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 728312070255
    Firmware version: 05.08.15.00

```

The line calling `rs2_pipeline_start_with_config` is the culprit.
Before pulling, the rs-color example worked, but I don't know which version that was. Probably a month old.

How to debug?

dorodnic, 2018-04-12:
Hi @kevindehecker 
Could you please call `export LRS_LOG_LEVEL="DEBUG"` to enable librealsense logs. The log file should be created in the running directory. Please share it. Also the `dmesg` log is always useful. 

If you have the option to find the version that worked it would help a lot. C samples were first introduced in v2.8.1. Seeing it work with different software would rule out all other causes.

kevindehecker, 2018-04-12:
I will attempt to figure out the version. It probably is still in the git history somewhere. In the mean time:

Reran with lrs_log_level debug, started the rs-color (which got stuck again), counted to 20, ctrl - c...
The log is empty, 0 bytes.

dmesg output over a usb re-plug:
```
[ 4578.597095] [c1] usb 4-1.1: USB disconnect, device number 4
[ 4582.766939] [c1] usb 4-1.1: new SuperSpeed USB device number 5 using xhci-hcd
[ 4582.787335] [c1] usb 4-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 4582.814627] [c1] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 4582.820967] [c1] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4582.828708] [c1] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 4582.836421] [c1] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 4582.844110] [c1] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 4582.852153] [c1] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 4582.862134] [c1] uvcvideo: Unable to create debugfs 4-5 directory.
[ 4582.867712] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 4582.877076] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[ 4582.885556] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 4582.895704] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 4582.905886] [c1] input: Intel(R) RealSense(TM) 430 as /devices/usb.1/12000000.dwc3/xhci-hcd.2.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input13
[ 4582.921300] [c1] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 4582.927674] [c1] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 4582.936311] [c4] uvcvideo: Unable to create debugfs 4-5 directory.
[ 4582.942640] [c4] uvcvideo 4-1.1:1.3: Entity type for entity Processing 7 was not initialized!
[ 4582.950322] [c4] uvcvideo 4-1.1:1.3: Entity type for entity Extension 8 was not initialized!
[ 4582.958682] [c4] uvcvideo 4-1.1:1.3: Entity type for entity Camera 6 was not initialized!
```

I have trouble interpreting the dmsg log, as most of the output also seemed to happen when it did work. (and actually also on my laptop, which works with the D435 just fine).

kevindehecker, 2018-04-12:
The version that did (and still does) work: ba7505c
(I made one change to the code, added the pie linker flag for android 7 40d20068d166fee8a3d108fd5497168b34acebe6 )

kevindehecker, 2018-04-12:
dmesg log from working version:
```
[ 4582.766939] [c1] usb 4-1.1: new SuperSpeed USB device number 5 using xhci-hcd
[ 4582.787335] [c1] usb 4-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 4582.814627] [c1] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 4582.820967] [c1] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4582.828708] [c1] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 4582.836421] [c1] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 4582.844110] [c1] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 4582.852153] [c1] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 4582.862134] [c1] uvcvideo: Unable to create debugfs 4-5 directory.
[ 4582.867712] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 4582.877076] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[ 4582.885556] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 4582.895704] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 4582.905886] [c1] input: Intel(R) RealSense(TM) 430 as /devices/usb.1/12000000.dwc3/xhci-hcd.2.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input13
[ 4582.921300] [c1] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 4582.927674] [c1] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 4582.936311] [c4] uvcvideo: Unable to create debugfs 4-5 directory.
[ 4582.942640] [c4] uvcvideo 4-1.1:1.3: Entity type for entity Processing 7 was not initialized!
[ 4582.950322] [c4] uvcvideo 4-1.1:1.3: Entity type for entity Extension 8 was not initialized!
[ 4582.958682] [c4] uvcvideo 4-1.1:1.3: Entity type for entity Camera 6 was not initialized!
[ 5478.910008] [c1] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 5478.916406] [c1] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 5478.924150] [c1] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 5478.931900] [c1] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 5478.939629] [c1] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 5478.947546] [c1] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 5478.957751] [c1] uvcvideo: Unable to create debugfs 4-5 directory.
[ 5478.963345] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 5478.972694] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[ 5478.981174] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 5478.991333] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 5479.001569] [c1] input: Intel(R) RealSense(TM) 430 as /devices/usb.1/12000000.dwc3/xhci-hcd.2.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input14
[ 5479.085861] [c2] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 5479.092259] [c2] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 5479.099972] [c2] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 5479.107624] [c2] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 5479.115346] [c2] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 5479.123523] [c2] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 5479.133733] [c2] uvcvideo: Unable to create debugfs 4-5 directory.
[ 5479.139494] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 5479.148695] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[ 5479.157166] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 5479.167320] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 5479.177575] [c2] input: Intel(R) RealSense(TM) 430 as /devices/usb.1/12000000.dwc3/xhci-hcd.2.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input15
[ 5479.245935] [c0] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 5479.252411] [c0] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 5479.260066] [c0] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 5479.267712] [c0] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 5479.275686] [c0] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 5479.283503] [c0] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 5479.293437] [c0] uvcvideo: Unable to create debugfs 4-5 directory.
[ 5479.299049] [c0] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 5479.308470] [c0] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[ 5479.316909] [c0] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 5479.327023] [c0] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 5479.337445] [c0] input: Intel(R) RealSense(TM) 430 as /devices/usb.1/12000000.dwc3/xhci-hcd.2.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input16
[... cut ...]
[  139.235611] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[  139.245012] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[  139.253511] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[  139.263677] [c2] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[  139.273859] [c2] input: Intel(R) RealSense(TM) 430 as /devices/usb.1/12000000.dwc3/xhci-hcd.2.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input7
```

dorodnic, 2018-04-12:
I have two suspicious PRs in that time range. Are you using the default V4L2 backend or the libuvc backend (by building with `-DFORCE_LIBUVC=true`)?
What exactly happens when you run the example? Is it stuck? How did you determine `rs2_pipeline_start_with_config` is involved?

kevindehecker, 2018-04-12:
Yes, I'm using libuvc, but that's because it is the default apparently. (never used the force flag for that). I actually get a cmake warning about using libuvc.

I determined the culprit line by adding some printf's, and it never passed through that command.

kevindehecker, 2018-04-12:
https://github.com/IntelRealSense/librealsense/blob/2f6a26aa784cc140f12b9144fdc2fb2171d410fa/CMakeLists.txt#L76

kevindehecker, 2018-04-12:
I tried to run rs-color rebuilded with force libuvc off, it still gets stuck.

I did notice a new line tailing in the demsg output:

```
[  224.155523] [c1] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[  224.165697] [c1] input: Intel(R) RealSense(TM) 430 as /devices/usb.1/12000000.dwc3/xhci-hcd.2.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input7
[  276.532133] [c2] xhci-hcd xhci-hcd.2.auto: Timeout while waiting for stop endpoint command
[  310.774649] [c0] set_aud_pll_rate[51] : aud_pll set_rate=180633600, get_rate = 180633609
[  314.777668] [c2] type=1400 audit(1523561167.706:14): avc: denied { getattr } for pid=5678 comm="top" path="/proc/iomem" dev="proc" ino=4026537599 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:proc_iomem:s0 tclass=file permissive=1

```

kevindehecker, 2018-04-14:
Affirmative, indeed the problem's cause is narrowed between 1def32af85e93e8332d633fd34eef10b144afb7c and 7c978164fba672036d1ebba1c107d2d5a97669f6
I have attempted to run gdb on rs-color, but unfortunately I have no idea how to do that on c++ code in Android...  Luckily, if I compile the code on my laptop (Ubuntu 16.04) with `-DFORCE_LIBUVC=true`, the problem can be reproduced!

It seems `thread.join` blocks in the libuvc deconstructor. https://github.com/IntelRealSense/librealsense/blob/2f6a26aa784cc140f12b9144fdc2fb2171d410fa/src/libuvc/libuvc.cpp#L290

This is probably caused by a problem in `power_D0()` when calling `uvc_open2`, which returns `UVC_ERROR_BUSY` which is not properly handled. 
https://github.com/IntelRealSense/librealsense/blob/2f6a26aa784cc140f12b9144fdc2fb2171d410fa/src/libuvc/libuvc.cpp#L378

If I apply the following "patch" 0f2b8af78248e11aaab4684072e1c26669888754 it solves the problem on the Odroid: 

```
odroidxu3:/data/lrs_binaries # ./rs-color                                                           
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 728312070255
    Firmware version: 05.08.15.00

RGB frame arrived.
First 10 bytes: 0e 10 0d 0f 12 0e 10 11 0c 0f 
Frame No: 1
Timestamp: 4960.253000
Timestamp domain: Hardware Clock
Time of arrival: 1523714760197

RGB frame arrived.
First 10 bytes: 0e 14 0d 0d 13 0c 11 12 0d 13 
Frame No: 3
Timestamp: 5027.453000
Timestamp domain: Hardware Clock
Time of arrival: 1523714760264

RGB frame arrived.
First 10 bytes: 90 88 90 92 8a 91 93 86 8d 91 
Frame No: 5
Timestamp: 5094.653000
Timestamp domain: Hardware Clock
Time of arrival: 1523714760329

^C
```

New problems:

1. I can only run it once. After ctrl-c, and restarting rs-color I get:

```                                                    
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 728312070255
    Firmware version: 05.08.15.00

rs_error was raised when calling rs2_pipeline_wait_for_frames(pipe:0xb6e1a648):
    Frame didn't arrived within 5000
```

2. If the Realsense is connected during a reboot, the device is not detected at all. Rebooting again without the realsense and connecting after reboot and rs-color works again (well , again only the one time...). 
3. Probably unrelated: my laptop does not like the rest of the libuvc implementation and crashes with a segfault

Regarding 1 and 2, these were already happening before when I was on the months old branch. (and the reason why I pulled master in the first place). Getting an Odroid in your lab might be a good idea then ;) . Although I have to admit it does work better with the Ubuntu install instead of Android. So, I think I will be switching to Ubuntu again. Still if you want to make a PR of this patch let me know.

dorodnic, 2018-04-16:
If I recall correctly items 1 and 2 are known issues at firmware level. I can suggest updating camera firmware using the following tool: [downloadcenter.intel.com/download/27514](https://downloadcenter.intel.com/download/27514?v=t), but I am not 100% certain that it was already resolved.

> my laptop does not like the rest of the libuvc implementation and crashes with a segfault
What do you mean?

kevindehecker, 2018-04-16:
> I can suggest updating camera firmware

Gave that a quick try, partial success. Running rs-color multiple times is still a no go, still resulting in the frame time out for all attempts after a first initial successfull run. However, booting with the realsense connected now seems to work, it is detected properly. 

>> my laptop does not like the rest of the libuvc implementation and crashes with a segfault

> What do you mean?

I mean that when I run the hacked 0f2b8af78248e11aaab4684072e1c26669888754 version of rs-color with force libuvc on, on my laptop Ubuntu 16.04, it segfaults out on me. All is fine V4L2, so not really a concern ATM.

```
There are 2 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 728312070255
    Firmware version: 05.09.02.00

Segmentation fault
```

kevindehecker, 2018-04-16:
Interesting development. After running the rs-color on the odroid, I let it sit for a while and then try it again. It now tells met there are 0 Realsense devices connected... I tried replugging -> no result. This is where it really gets interesting: I plugged a usb mouse in the usb port previously used for the Realsense. The mouse is not detected also, also not in the other usb3 port! The usb3 is totally out. Rebooting worked in the sense the usb3 ports worked once again.

kevindehecker, 2018-04-24:
I've moved to Ubuntu, and since a work around patch is provided above I'll close this issue.

closed 2018-04-24 by kevindehecker