realsenseai/librealsense #1536: macOS Installation issues

orgicus, 2018-04-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 | D435 } | 
| Firmware Version                   | D415: 05.09.02.00 | D435: 05.08.14.00) | 
| Operating System & Version |   MacOS 10.11.6  | 
| Platform                                 | Macbook Pro  |
| SDK Version                            |  { 2.10.3 }                          | 

### macOS Installation issues

I ran into multiple issues simply following the [macOS installation doc](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md):

```bash
	Undefined symbols for architecture x86_64:
	  "librealsense::make_less_screamy(char const*)", referenced from:
	      librealsense::get_string(rs2_rs400_visual_preset) in rs_advanced_mode.cpp.o
	  "_libusb_error_name", referenced from:
	      librealsense::platform::libuvc_backend::query_usb_devices() const in libuvc.cpp.o
	      librealsense::platform::libuvc_uvc_device::get_data_usb(uvc_req_code, int, int) const in libuvc.cpp.o
	      librealsense::platform::libuvc_uvc_device::set_data_usb(uvc_req_code, int, int, int) const in libuvc.cpp.o
	      librealsense::platform::uvclib_usb_device::uvclib_usb_device(librealsense::platform::usb_device_info const&) in libuvc.cpp.o
	      librealsense::platform::uvclib_usb_device::foreach_usb_device(libusb_context*, std::__1::function<void (librealsense::platform::usb_device_info const&, libusb_device*)>) in libuvc.cpp.o
	      librealsense::platform::uvclib_usb_device::send_receive(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&, int, bool) in libuvc.cpp.o
	  "_libusb_get_device_speed", referenced from:
	      librealsense::platform::get_usb_descriptors(libusb_device*) in libuvc.cpp.o
	  "_libusb_get_parent", referenced from:
	      librealsense::platform::uvclib_usb_device::foreach_usb_device(libusb_context*, std::__1::function<void (librealsense::platform::usb_device_info const&, libusb_device*)>) in libuvc.cpp.o
	  "_libusb_get_port_numbers", referenced from:
	      librealsense::platform::get_usb_descriptors(libusb_device*) in libuvc.cpp.o
	  "_libusb_handle_events_completed", referenced from:
	      _uvc_handle_events(void*) in init.cpp.o
	      void* std::__1::__thread_proxy<std::__1::tuple<uvc_start_handler_thread(uvc_context*)::$_0> >(void*) in init.cpp.o
	ld: symbol(s) not found for architecture x86_64
	```

I already had libusb and libuvc installed on my system so the fix was to set `USE_SYSTEM_LIBUSB = OFF`

which left me with only one error to deal with:

```bash
	Undefined symbols for architecture x86_64:
	  "librealsense::make_less_screamy(char const*)", referenced from:
	      librealsense::get_string(rs2_rs400_visual_preset) in rs_advanced_mode.cpp.o
	ld: symbol(s) not found for architecture x86_64
	clang: error: linker command failed with exit code 1 (use -v to see invocation)
	make[2]: *** [librealsense2.2.10.3.dylib] Error 1
	make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
	make: *** [all] Error 2
```

...for which I just used a very hacky workaround:

```c++
	#define STRCASE(T, X) case RS2_##T##_##X: {\
        static std::string s##T##_##X##_str = (#X);\
        return s##T##_##X##_str.c_str(); }
    ``` in `src/ds5/advanced_mode/rs_advanced_mode.cpp`

Please let me know:

1. What's the (clean) recommended way of dealing with the `make_less_screamy` error
2. What's the best way of contributing to save others from this trouble (e.g. update [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md), [Q&A](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A) or submit a pull request) ?

Thank you,
George

dorodnic, 2018-04-15:
Hi George,
Sorry the installation didn't go smoothly.
Regarding the first issue, you are correct that we do not address it in any of our documentation. The document assumes clean system. Adding a paragraph about `USE_SYSTEM_LIBUSB` would be a great contribution, if you would be interested. We are very open to community [contributions](https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md).
Regarding the second issue, it seems a bit strange. We run regular Mac OS builds ([latest here](https://travis-ci.org/IntelRealSense/librealsense/jobs/364984524)) to make sure everything compiles out of the box. Are you enabling any auxiliary CMake flags (beyond `-DBUILD_EXAMPLES=true`). Perhaps there is older version of librealsense2 **installed** on your system?

orgicus, 2018-04-15:
@dorodnic You are totally right! 

1. I'll read the contributions guidelines.
2. It's amazing you caught the older version of librealsense as an issue: I totally forgot I played with librealsense (1) a couple of summers ago. I had a clone of librealsense at commit `05f0474b9fa9305e9c4528c88f19e110b37d5e4e` (Wed Jun 29 13:58:29 2016 -0700). which was already compiled and installed on the system. Perhaps I should add a not on the Q&A trouble shooting wiki page ?

dorodnic, 2018-04-15:
Great!
We are trying to automate validation as much as possible, but when building from source there are so many things that can go wrong. Not everything we can always reproduce, so we tried to make contributing as easy as possible.

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
let me close this ticket, and please feel free to open new ticket if any question.

closed 2018-05-11 by RealSense-Customer-Engineering