realsenseai/librealsense #1537: IR channels damaged after using Customcalibration tool

BGA3, 2018-04-13:
Hi

I have used the CustomCalibration tool to calibrate the D415 camera, and afterwards used the CustomRW tool to configure the camera (both in version 2.5.2.0).
The before- and after configuration are just minor changed. 

But now, the IR channels are destroyed and consequently the depth map. It seems that the IR image is constructed wrongly. When using the Realsense Viewer I get alot of "unknown errors" (great). See screenshot here:
https://imgur.com/a/L9nkZ

What is this issue, and can someone help resolve it? Btw I have tried "hardware reset", without any luck.

RealSense-Customer-Engineering, 2018-04-19:
[Realsense Customer Engineering Team Comment] 
hi @BGA3,
can you please try if customerRW tool with "-g" to restore the calibration?
let us know if it works there, thanks.

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
hi @BGA3,

do you need more help on this topic?
thanks.