realsenseai/librealsense #1540: There are 0 connected RealSense devices.

luanps, 2018-04-14:
When I run rs-depth or another example I've got 
> There are 0 connected RealSense devices.

and rs-enumerate-devices
> No device detected. Is it plugged in?

I've already tried [patch installation](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md) and [dkms](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) version ( one at time). Both installation run without errors. 
I'm using a built-in webcam on Dell Inspiron 15 5557

Distributor ID:	Ubuntu 
Description:	Ubuntu 16.04.4 LTS (fresh installation)
Release:	16.04
Codename:	xenial
4.13.0-36-generic

*by default, I can access depth camera only using cheese.

dorodnic, 2018-04-15:
Ho @luanps 
What is the output of `lsusb` and `dmesg`?

closed 2018-05-01 by luanps

ssakhavi, 2018-11-02:
@JasonChang13 

I have a similar problem.

This is the message for dmesg:
tissot:/ $ cd data
tissot:/data $ ls
ls: .: Permission denied
1|tissot:/data $ su
tissot:/data # ls
FTM_AP
adb
anr
app
app-asec
app-ephemeral
app-lib
app-private
backup
bootchart
cache
connectivity
dalvik-cache
data
dpm
drm
fota
hostapd
lineageos_updates
local
lost+found
lrs_binaries
media
mediadrm
misc
misc_ce
misc_de
nfc
ota
ota_package
property
resource-cache
shared
ss
ssh
stock_boot_95025733fe36dabfbc8f7f754db060abf4773398.img.gz
system
system_ce
system_de
time
tombstones
user
user_de
vendor
tissot:/data # cd lrs_binaries/
/rs-enumerate-devices                                     <
 02/11 02:44:10,500 ERROR [-182590300] (types.h:179) fail to initialize libuvc Last Error: No such file or directory
terminating with uncaught exception of type librealsense::linux_backend_exception: fail to initialize libuvc Last Error: No such file or directory
/usr/local/google/buildbot/src/android/ndk-release-r18/external/libcxx/../../external/libcxxabi/src/abort_message.cpp:73: abort_message: assertion "terminating with uncaught exception of type librealsense::linux_backend_exception: fail to initialize libuvc Last Error: No such file or directory" failed
Aborted
hmod -R 777 /dev/bus/usb/*                                <
chmod: /dev/bus/usb/*: No such file or directory
hmod -R 777 /dev/bus/usb/                                 <
chmod: /dev/bus/usb/: No such file or directory
hmod -R 777 /dev/bus/usb                                  <
chmod: /dev/bus/usb: No such file or directory
hmod -R 777 /dev/bus/usb                                  <
/rs-enumerate-devices                                     <
No device detected. Is it plugged in?
/rs-enumerate-devices                                     <
[... 951 lines cut ...]
[   80.191880] fpc1020 soc:fpc1020: fpc1020_irq_handler
[   80.196781] fpc1020 soc:fpc1020: fpc1020_irq_handler
[   80.447830] max98927_reg_put: register 0x40, value 0x10
[   80.447914] max98927_spk_gain_put: 5
[   80.448112] max98927_digital_gain_put: 56
[   80.448310] max98927_left_channel_enable_set: register 0xFF, value 0x01
[   80.450147] type=1400 audit(1541140984.884:21): avc: denied { search } for pid=580 comm="android.hardwar" name="default" dev="tmpfs" ino=5490 scontext=u:r:hal_fingerprint_default:s0 tcontext=u:object_r:storage_file:s0 tclass=dir permissive=0
[   80.450161] type=1400 audit(1541140985.674:22): avc: denied { search } for pid=503 comm="writer" name="/" dev="mmcblk0p27" ino=2 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:persist_file:s0 tclass=dir permissive=0
[   80.451910] ------max98927_dai_hw_params------
[   80.452127] max98927_dai_hw_params: format supported 2
[   80.452867] ------max98927_set_clock------
[   80.460351] max98927_stream_mute--- stream 0, mute 0
[   81.247396] binder_alloc: 2171: binder_alloc_buf size 1056768 failed, no address space
[   81.247414] binder_alloc: allocated: 6640 (num: 12 largest: 2104), free: 1032336 (num: 4 largest: 1028832)
[   81.247426] binder: 2814:2825 transaction failed 29201/-28, size 1056768-0 line 3056
[   81.247446] binder: send failed reply for transaction 85001 to 2171:2767
[   81.275441] binder_alloc: 2171: binder_alloc_buf size 1056768 failed, no address space
[   81.275458] binder_alloc: allocated: 6640 (num: 12 largest: 2104), free: 1032336 (num: 4 largest: 1028832)
[   81.275468] binder: 2814:2825 transaction failed 29201/-28, size 1056768-0 line 3056
[   81.275484] binder: send failed reply for tra

This is the message from lsusb:

Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 001: ID 1d6b:0003