realsenseai/librealsense #1541: How could I install the SDK on Travis CI?

Gnimuc, 2018-04-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | NA | 
| Firmware Version                   | NA | 
| Operating System & Version |   Ubuntu 14(Travis CI)  | 
| Kernel Version (Linux Only)    |  4.4.0-101                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.x                          | 

### Issue Description
I'm writing a Julia wrapper and want to set up a CI environment. Following this [doc](https://github.com/IntelRealSense/librealsense/blob/v2.10.3/doc/distribution_linux.md), I failed to install the SDK on Travis CI. I know the doc says the distribution package is for Ubuntu 16, so the errors below might be expected. But is there any workaround for install the SDK without building it from scratch(it's very time consuming)? Any help would be highly appreciated. :)

https://travis-ci.org/Gnimuc/RealSense.jl/jobs/366487987:
```
Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.2.0, 4.4.0-101-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.2.0
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
Setting up librealsense2-udev-rules:amd64 (2.10.3-0~realsense0.66) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 librealsense2-utils : Depends: librealsense2 (= 2.10.3-0~realsense0.66) but it is not going to be installed
```

dorodnic, 2018-04-15:
I suspect the `dkms` package is what causing the problem. Perhaps what you could do is `wget` the `deb` files directly:
```cpp
http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/pool/xenial/main/libr/librealsense2/librealsense2-dev_2.10.3-0~realsense0.66_amd64.deb

http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/pool/xenial/main/libr/librealsense2/librealsense2-utils_2.10.3-0~realsense0.66_amd64.deb

http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/pool/xenial/main/libr/librealsense2/librealsense2_2.10.3-0~realsense0.66_amd64.deb
```
And install them directly using `dpkg`.
We need to design a more conventional solution for this problem, but until then I hope this could work for you.

Gnimuc, 2018-04-15:
Hi @dorodnic, thanks for the quick reply. I tried to install the `deb` files directly, but it complains `Package libglfw3 is not installed.` I had built, installed and `ldconfig`ed libglfw before installing the `deb`. I'm wondering what's the correct path that `librealsense2_2.10.3-0~realsense0.66_amd64.deb` was looking for.
```
-- Installing: /usr/local/lib/libglfw.so.3.1
-- Installing: /usr/local/lib/libglfw.so.3
-- Installing: /usr/local/lib/libglfw.so
```

```
Saving to: ‘librealsense2_2.10.3-0~realsense0.66_amd64.deb’
100%[======================================>] 1,659,058    668KB/s   in 2.4s   
2018-04-15 13:06:24 (668 KB/s) - ‘librealsense2_2.10.3-0~realsense0.66_amd64.deb’ saved [1659058/1659058]
Selecting previously unselected package librealsense2:amd64.
(Reading database ... 99680 files and directories currently installed.)
Preparing to unpack librealsense2_2.10.3-0~realsense0.66_amd64.deb ...
Unpacking librealsense2:amd64 (2.10.3-0~realsense0.66) ...
dpkg: dependency problems prevent configuration of librealsense2:amd64:
 librealsense2:amd64 depends on libglfw3; however:
  Package libglfw3 is not installed.
dpkg: error processing package librealsense2:amd64 (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.19-0ubuntu6.13) ...
Errors were encountered while processing:
 librealsense2:amd64
```

dorodnic, 2018-04-15:
That's strange. I was under impression that `librealsense` was not dependent on GLFW and only `librealsense-utils` was. @ev-mp, do you have any idea why it fails?

Gnimuc, 2018-04-15:
Here is my current travis configuration:
```yaml
before_script:
 - if [ $TRAVIS_OS_NAME = linux ]; then
     sudo add-apt-repository --yes ppa:ubuntu-toolchain-r/test;
     sudo apt-get update -qq;
     sudo apt-get install -qq xorg-dev;
     sudo apt-get install -qq gcc-5 g++-5;
     sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 60 --slave /usr/bin/g++ g++ /usr/bin/g++-5;
     wget https://github.com/glfw/glfw/releases/download/3.1.1/glfw-3.1.1.zip;
     unzip glfw-3.1.1.zip && cd glfw-3.1.1;
     cmake -G "Unix Makefiles" -DBUILD_SHARED_LIBS=ON;
     make;
     sudo make install && ldconfig && cd ..;
     sudo ln -s /usr/local/lib/libglfw.so /usr/lib/x86_64-linux-gnu/libglfw.so

     wget http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/pool/xenial/main/libr/librealsense2/librealsense2_2.10.3-0~realsense0.66_amd64.deb;
     sudo dpkg -i librealsense2_2.10.3-0~realsense0.66_amd64.deb;
     wget http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/pool/xenial/main/libr/librealsense2/librealsense2-dev_2.10.3-0~realsense0.66_amd64.deb;
     sudo dpkg -i librealsense2-dev_2.10.3-0~realsense0.66_amd64.deb;
   fi
```

dorodnic, 2018-04-18:
Hi @Gnimuc 
Could you please try your original approach (`sudo apt-get install librealsense-dev`). We made some changes that should help and published new packages, but I didn't have time to test with your Travis settings.

Gnimuc, 2018-04-18:
Unfortunately, the build still fails...
```
Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.2.0, 4.4.0-101-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.2.0
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
Setting up librealsense2-udev-rules:amd64 (2.10.4-0~realsense0.67) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 librealsense2-dev : Depends: librealsense2 (= 2.10.4-0~realsense0.67) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

## .travis.yml
```yml
before_script:
 - if [ $TRAVIS_OS_NAME = linux ]; then
     echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list;
     sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE;
     sudo apt-get update -qq;
     sudo apt-get install linux-headers-$(uname -r) -y;
     sudo apt-get install librealsense2-dkms -y;
     sudo apt-get install librealsense2-dev -y;
   fi
```

Gnimuc, 2018-04-18:
Hi @dorodnic, I just tried installing the dkms package directly and it's working! Thanks!

```
http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/pool/xenial/main/libr/librealsense2/librealsense2-dev_2.10.4-0~realsense0.67_amd64.deb

http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/pool/xenial/main/libr/librealsense2/librealsense2_2.10.4-0~realsense0.67_amd64.deb
```