realsenseai/librealsense #1542: Enable/Disable auto exposure C++ API

vladsterz, 2018-04-14:
*---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D435  | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC|

Title says it all. How do I access this functionality with C++ API.

zivsha, 2018-04-14:
Take a look at sensor-control example, specifically:

https://github.com/IntelRealSense/librealsense/blob/2f6a26aa784cc140f12b9144fdc2fb2171d410fa/examples/sensor-control/api_how_to.h#L147-L184

vladsterz, 2018-04-15:
Thanks alot!

closed 2018-04-15 by vladsterz