realsenseai/librealsense #1545: Potential Python binding issue

jimmy82, 2018-04-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300                                                     | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 14/16/17) / MacOS        | 
| Kernel Version (Linux Only)    |  4.14.13                                                  | 
| Platform                                 | PC                                                          |
| SDK Version                            |  2.10.3                                                   | 

### Issue Description
I was trying to call the 'seek' function in rs2::playback to control the playback of .bag files.

However, PyBind issue a critical error as it was unable to translate the float parameter into the Chrono object. It suggested me to re-check the compilation of the PyBind headers.

Is this implementation incomplete? Alternative, is there an easier way to seek a .bag file and also be notified at the end of the playback?

Thanks.

jimmy82, 2018-04-16:
Hi, thanks for the support.

Is there a Python example where I can use a playback device context to read all video frames without frame drop?

dorodnic, 2018-04-16:
Hi @jimmy82 
At the moment, no, but we will try to add such an example (@lramati / @AnnaRomanov could you please assist?)

dorodnic, 2018-04-18:
I'm closing the `seek` issue. We are looking into adding the python code snippet you requested. You can open a separate issue on this for follow-up if you'd like.

closed 2018-04-18 by dorodnic

jimmy82, 2018-04-18:
Hi,

    Thanks. It will be good if Python documentation can be beef up.

soarwing52, 2019-09-04:
Hi, I haven't found any functioning example of playback.seek in python

I can get playback. [current_status,file_name,get_duration,get_position,pause, resume, set_real_time] all working

but seek still throws back error

`TypeError: seek(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.playback, Sets the playback to a specified time point of the played data.time: datetime.timedelta) -> None

Invoked with: <pyrealsense2.device: Intel RealSense D435 (S/N: 817512071792)>, 0`

in which the type of my playback is <class 'pyrealsense2.pyrealsense2.playback'>
it should be correct, I would like to ask if you can also answer this 

thank you

lramati, 2019-09-04:
Can you please open a new issue for your problem and indicate when version of the SDK you are using and what the line you tried to run is?

PR #1550 (2018-04-16): Add support for conversions between C++ and Python time objects
Add support for conversions between C++ std::chrono types and Python datetime types to fix #1545.

In python, `pyrealsense2.playback.seek()` should now accept `datetime.timedelta`s and `float`s as parameters.
Additionally, `pyrealsense2.playback.get_duration()` now returns a native python `datetime.timedelta` object