realsenseai/librealsense #1548: RGB and Depth frames not synchronized

vladsterz, 2018-04-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D435  | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC|
| SDK Version                            |   2.10.3                         | 
|Firmware Version                    | 05.09.02.00 |

### Issue Description
Recently I upgraded the FW version and I was told that turning the Auto Exposure off will synchronize my RGB and Depth streams. I tried this ( using [this](https://github.com/IntelRealSense/librealsense/issues/1542) info ) and tried 3 different ways to acquire frames.  So far I've tried i) **wait_for_frames()** , ii) **poll_for_frames()** and iii) **asynchronous_syncher** and I am testing if the frames are synchronized like this:
`auto rgb_t = frameset.get_color_frame().get_timestamp();`
`auto d_t = frameset.get_depth_frame().get_timestamp();`
`std::cout<< rgb_t - d_t <<std::endl;`

The results with every one of these ways looks something like this:
[image]
 My question is: Am I missing something or doing something wrong measuring the time? Or is it a bug?

vladsterz, 2018-04-20:
Any help?

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@vladsterz
The synchronization between two streams from the same sensor like depth and IR should be perfectly synchronized by frame number.
Synchronization between two streams from separated sensors, like color and depth, is achieved by best match of the frames timestamp,
It’s expected that frames within a synchronized frameset will have a temporal offset, bounded by a period of one  frame with the lowest FPS
The above assumes that there are no frame drops.

vladsterz, 2018-05-15:
Hello @RealSense-Customer-Engineering ,

> Synchronization between two streams from separated sensors, like color and depth, is achieved by best match of the frames timestamp,

Do I have to do it myself then? Do I have to explicitly choose the frames have the smallest temporal difference?

vladsterz

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@vladsterz
no, the wait_for_frames has done the best mapping for you.

soarwing52, 2019-04-04:
In the Python wrapper of the latest version,
pipeline = rs.pipeline()
config = rs.config()
config.enable_device_from_file(file_name)
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgba8, 30)
# Start streaming from file
profile = pipeline.start(config)

and then use 
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)
        depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()
        time_color = rs.frame.get_timestamp(color_frame)
        time_depth = rs.frame.get_timestamp(depth_frame)

the actual frames I recorded has no frame drop, but when I read the bad file, the time gap can be up to 10000 or more, not so stable and made my measurement of 3D distance inccorecet 

I wonder is it a problem with wait for frames or I need to use other method.
currently I use a loop after getting the depth frame and then if the gap is bigger than 20, I wait for frames again and get color frame to find the real matching frame.

tispratik, 2020-03-16:
I could fix the time gap issue by installing librealsense from source and enabling meta data. https://dev.intelrealsense.com/docs/build-configuration