realsenseai/librealsense #1553: Realsense Viewer doesn't recognize camera d435

rafaor29, 2018-04-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 |
| Operating System & Version | Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.13.0-38-generic  | 
| Platform                                 | PC |
| SDK Version                            | 2.10.3  | 

### Issue Description
When i initialize the realsense viewer it doesn't recognize the camera even though the computer does. I found out that it is a kernel problem, but i'd like to work with this one.
In the next image you can see that the computer detects the camera:
[image]

But it doesn't recognize it in the realsense viewer:
[image]

What can i do to fix this? I followed all the steps to install the SDK properly.

rafaor29, 2018-04-17:
Hi @ev-mp , i tried what you said but still not working. 
[image]
[image]
I also tried to disable the secure boot by doing: 
`sudo apt install mokutil`
`sudo mokutil --disable-validation`
But still not working. 
I downloaded the github and followed the instructions of the readme.

RealSense-Customer-Engineering, 2018-04-19:
[Realsense Customer Engineering Team Comment] 
Hi @rafaor29,

Please check if the uvc kernel module there. You can also run "modinfo uvcvideo | grep ver" to show the kernel module info.
/lib/modules/4.13.0-38-generic/kernel/drivers/media/usb/uvc/uvcvideo.ko

Wampuscat, 2018-04-23:
Have you tried updating the firmware on the camera?  This fixed a similar issue for me.  Good Luck.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hi @rafaor29,

Did you resolve the issue? Did you also try different kernel like 4.10?

RealSense-Customer-Engineering, 2018-06-04:
[Realsense Customer Engineering Team Comment]
Hi @rafaor29,

Any update?

GuidarelliG, 2018-09-23:
Hello, 

I am not sure If this will get to anyone but I am having the same issue.
staring from the first @RealSense-Customer-Engineering comment I have:

gabriel@gabriel-K501UW:~$ modinfo uvcvideo | grep ver

version:        1.1.2.realsense-1.3.1
description:    USB Video Class driver
srcversion:     1E90D74E555F0FCE05F7841
vermagic:       4.4.0-134-generic SMP mod_unload modversions retpoline