realsenseai/librealsense #1556: Align infrared frame to color

Panepo, 2018-04-17:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version | Win10 (Build 16299)  | 
| Platform                                 | PC |
| SDK Version                            | 2.10.0                          | 

Hi there,

I am trying to align infrared frame to color but get these error message:
```
RealSense error calling rs2_get_frame_width(frame_ref:nullptr):
    null pointer passed for argument "frame_ref"
```
and a part of my code is
```
  rs2::align alignTo(RS2_STREAM_COLOR);
  rs2::frameset data = pipeline.wait_for_frames();
  rs2::frameset alignedFrame = alignTo.process(data);
  
  rs2::frame frameColor = alignedFrame.get_color_frame();
  rs2::frame frameInfrared = alignedFrame.get_infrared_frame();
  rs2::depth_frame depth = alignedFrame.get_depth_frame();
```
This code works only if the align is set to `RS2_STREAM_DEPTH`

Is it possible to align infrared frame to color ? or is there something wrong on my code ?

Thanks

RealSense-Customer-Engineering, 2018-04-27:
[Realsense Customer Engineering Team Comment]

@Panepo
The align is only performed in between depth and other image. you can refer to the API document description.

inline

Create align processing block Alignment is performed between a depth image and another image. To perform alignment of a depth image to the other, set the align_to parameter with the other stream type. To perform alignment of a non depth image to a depth image, set the align_to parameter to RS2_STREAM_DEPTH Camera calibration and frame's stream type are determined on the fly, according to the first valid frameset passed to process() (file:///C:/Users/cpai/Documents/Visual%20Studio%202015/Projects/latestLibRS/librealsense-2.10.4/doc/doxygen/html/classrs2_1_1align.html#abb46dd776cc14b1dd8d6bbf371dd72a0)

Panepo, 2018-04-30:
Hi @RealSense-Customer-Engineering,

We are trying to build our face recognition algorithm by D415, If the color and infrared frames are aligned would be helpful. 

If set the set the align_to parameter to `RS2_STREAM_DEPTH` causes the RGB and infrared images zoomed out and broken if there is no depth data. Or maybe there is an option to make RGB and infrared frame keep the same resolution and quality ?

Thanks

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@Panepo
The `align` class used in librealsense demos maps between depth and some other stream and vice versa. We do not offer other forms of stream alignments.

But one suggestion for you to have a try, Basically the mapping is a triangulation technique where we go through the intersection point of a pixel in 3D space to find its origin in another frame, this method work properly when the source data is depth (Z16 format). One possible way to map between two none-depth stream is to play three streams (Depth+IR+RGB), then calculate the UV map for Depth to Color, and then use this UV map to remap IR frame ( remember that Depth and left IR are aligned by design).

Hope the suggestion give you some idea.

Panepo, 2018-05-16:
Hi @RealSense-Customer-Engineering,

Thanks a lot.

closed 2018-05-16 by Panepo

idoadse, 2021-01-21:
> [Realsense Customer Engineering Team Comment]
> @Panepo
> The `align` class used in librealsense demos maps between depth and some other stream and vice versa. We do not offer other forms of stream [...]

Could you elaborate on how to extract and apply the uv map and applying on the frames? 
I'm using pyrealsense2, and currently only succeeed to perform alignment **to depth** with the following code:

`    
    align_to = rs.stream.depth
    align = rs.align(align_to)
    aligned_frames = align.process(frames)

    aligned_infrared_frame = aligned_frames.get_infrared_frame()
    color_frame = aligned_frames.get_color_frame()
`