realsenseai/librealsense #1559: Binaries of the examples

manyaafonso, 2018-04-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 2.10.3 | 
| Operating System & Version |   Linux Mint 18.2  | 
| Platform                                 | PC |
| SDK Version                            |  2.10.3     | 

### Issue Description
I do not have the camera with me, I am processing images acquired with an SR300. I followed the instructions from distribution.md to install librealsense and i can run realsense-viewer from the terminal and could open the .bag file from the samples available from https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

Where in the build are the example binaries? I was expecting some executable files that can be called from the terminal. Or are the examples only for Windows?

dorodnic, 2018-04-17:
You can lunch the `realsense-viewer` and open the recordings that way. All the examples are also distributed under `librealsense2-utils`

manyaafonso, 2018-04-17:
I need to alter the code from the samples and then link it with librealsense, that is why I need to build from source.

manyaafonso, 2018-04-17:
The CMakeLists from examples/pointcloud contains the line
add_executable(rs-pointcloud rs-pointcloud.cpp ../example.hpp)
so my question is very simple, where is the executable rs-pointcloud?

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
Hi @manyaafonso,

If you build librealsense apps from the source code, the default installation path is in /usr/local, YOu should find those apps in /usr/local/bin.
if you install "librealsense2-utils" package, those apps should be in /usr/bin.
You can also define the flag "CMAKE_INSTALL_PREFIX" to assign the installation path you want.

manyaafonso, 2018-04-18:
Thanks, @RealSense-Customer-Engineering . I found them.

closed 2018-04-18 by manyaafonso