realsenseai/librealsense #1562: cs-tutorial-2-capture ''Frame didn't arrived within 5000 rs2_pipeline_wait_for_frames'

DvdZmd, 2018-04-17:
I can run successful the cs-tutorial-1-depth in VS 2017, but the cs-tutorial-2-capture only works fine the first time i run it after to start up the computer, but if i stop the debuging, later VS give me this exception if try to debug: 

System.Exception: 'Frame didn't arrived within 5000
rs2_pipeline_wait_for_frames(pipe:000000001CE9BA20)'

I was looking around in the net for a solution but no fix could help me, thats why i start a new thread.

Someone tell me that maybe is a power supply issue, but i don't think so beacuse the The RealSense Viewer works fine ever.

Also i was trying to add the two reference in a new proyect (i installed the lasted release of the SDK) as said in the github; for Intel.RealSense.dll there is no problem, but when i try to add realsense2.dll it returned:

realsense2.dll could not be added. Please make sure that the file is accesible, and that it is a valid assembly or COM component.

dorodnic, 2018-04-18:
Hi @DvdZmd 
`cs-tutorial-2-capture` should be functionally equivalent to `rs-capture`, did you try to run it? 
In the Viewer I would recommend to try higher resolutions of depth and color, to rule out hardware malfunction

DvdZmd, 2018-04-18:
Hi @dorodnic, thanks for the reply
I tryed the C++ example rs-capture and it work perfectly!
I will begin my develoment in C++ so.
Otherwise the Viewer with resolution 1280x720 work fine but if i switch to 1920x1080 (The D415 support that resolution for RGB) it doesn't work: "No frame recieved!"

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
@DvdZmd  Please note that for D415, up to 1280x720 active stereo depth resolution supported. That's why you see "No frame recieved!" when set depth to 1920x1080.

And considering the previous issue resolved, can we close this issue?

closed 2018-06-05 by RealSense-Customer-Engineering