realsenseai/librealsense #1563: **Illegal instruction (core dumped)** and **malfunction viewer**

a-loner, 2018-04-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.9.2.0 | 
| Operating System & Version | Ubuntu 16.04 | 
| Kernel Version (Linux Only)    | 4.13.0-38-generic | 
| Platform                                 | UP Squared (Model#: UPS-APLP4-A10-08128 )|
| SDK Version                            | 2.10.3 | 

### Initial Problem
At first, I tried to install the prebuilt package strictly following [this instruction](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). When I try to lunch the viewer as suggested. The error I got is shown as below.

```bash
Illegal instruction (core dumped)
```

### Attempts for solving the problem 

1. I upgraded the firmware to 5.9.2.0 using another desktop Windows PC. The problem is **still the same**. 
2. Then I tried to reinstall the OS and then build librealsense from source strictly following [this instruction](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). The problem is **still the same**.
3. Then I tried to **git checkout** to older version (**v2.10.1**) as suggested in issue [#1491](https://github.com/IntelRealSense/librealsense/issues/1491) and re-compile everything using following command. 
```bash
sudo make uninstall && make clean && make -j4 && sudo make install
```
I **could** launch the viewer, but there is **only one stream** I could choose from (i.e. Stereo) and it **cannot retrieve any frame** when I toggle it on.

### Then I tried several other version:
**v2.10.2** has same problem as v2.10.3, showing 
```bash
Illegal instruction (core dumped)
```

**v2.10.0** cannot compile, reporting error:

```bash
-- Info: REALSENSE_VERSION_STRING=2.10.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sys_er26/workspace/librealsense/build
-- Uninstalling "/usr/local/lib/librealsense2.so.2.10.2"
-- Uninstalling "/usr/local/lib/librealsense2.so.2"
-- Uninstalling "/usr/local/lib/librealsense2.so"
-- Uninstalling "/usr/local/include/librealsense2/rs_advanced_mode.h"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_context.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_frame.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_device.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_record_playback.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_internal.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_sensor.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_processing.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_pipeline.hpp"
-- Uninstalling "/usr/local/include/librealsense2/hpp/rs_types.hpp"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_context.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_record_playback.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_frame.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_types.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_device.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_option.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_advanced_mode_command.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_processing.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_internal.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_sensor.h"
-- Uninstalling "/usr/local/include/librealsense2/h/rs_pipeline.h"
-- Uninstalling "/usr/local/include/librealsense2/rsutil.h"
-- Uninstalling "/usr/local/include/librealsense2/rs_advanced_mode.hpp"
-- Uninstalling "/usr/local/include/librealsense2/rs.hpp"
-- Uninstalling "/usr/local/include/librealsense2/rs.h"
-- Uninstalling "/usr/local/lib/cmake/realsense2/realsense2Targets.cmake"
-- Uninstalling "/usr/local/lib/cmake/realsense2/realsense2Targets-release.cmake"
-- Uninstalling "/usr/local/lib/cmake/realsense2/realsense2Config.cmake"
[... 161 lines cut ...]
/usr/bin/ld: CMakeFiles/rs-measure.dir/rs-measure.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
examples/measure/CMakeFiles/rs-measure.dir/build.make:98: recipe for target 'examples/measure/rs-measure' failed
make[2]: *** [examples/measure/rs-measure] Error 1
CMakeFiles/Makefile2:645: recipe for target 'examples/measure/CMakeFiles/rs-measure.dir/all' failed
make[1]: *** [examples/measure/CMakeFiles/rs-measure.dir/all] Error 2
[ 72%] Building CXX object examples/post-processing/CMakeFiles/rs-post-processing.dir/__/__/third-party/imgui/imgui_draw.cpp.o
[ 72%] Building CXX object examples/post-processing/CMakeFiles/rs-post-processing.dir/__/__/third-party/imgui/imgui_impl_glfw.cpp.o
[ 72%] Building CXX object examples/align/CMakeFiles/rs-align.dir/__/__/third-party/imgui/imgui_draw.cpp.o
[ 73%] Building CXX object examples/align/CMakeFiles/rs-align.dir/__/__/third-party/imgui/imgui_impl_glfw.cpp.o
[ 74%] Linking CXX executable rs-post-processing
/usr/bin/ld: CMakeFiles/rs-post-processing.dir/rs-post-processing.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
examples/post-processing/CMakeFiles/rs-post-processing.dir/build.make:176: recipe for target 'examples/post-processing/rs-post-processing' failed
make[2]: *** [examples/post-processing/rs-post-processing] Error 1
CMakeFiles/Makefile2:865: recipe for target 'examples/post-processing/CMakeFiles/rs-post-processing.dir/all' failed
make[1]: *** [examples/post-processing/CMakeFiles/rs-post-processing.dir/all] Error 2
[ 75%] Linking CXX executable rs-align
[ 75%] Built target rs-align
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

```

dorodnic, 2018-04-18:
Hi @a-loner 
This issue is already resolved on the `development` branch and will be published as part of an official release shortly. 
Please try `git checkout development` before `cmake` and `make` steps.

a-loner, 2018-04-18:
Hi @dorodnic 

Thanks for sharing this information. Yes, I can confirm it's working now.

dorodnic, 2018-04-18:
The official [v2.10.4](https://github.com/IntelRealSense/librealsense/releases/tag/v2.10.4) with the fix is also out now. Thank you for reporting the issue

closed 2018-04-18 by dorodnic