realsenseai/librealsense #1564: intel realsense d435 and  labview

lonely-users, 2018-04-18:
Hello,

      I want to use Labview 2017 to develop Realsense d435 in Windows 10, I found a set of instanceshttps://github.com/IntelRealSense/librealsense/tree/development/wrappers/labview，but there is no way to use it after download, subVI is missing and the .dll file cannot be imported(missing corresponding header file).

     Who can give me some suggestions or tips?

     Thank you!
[image]
[image]
[image]

agrunnet, 2018-04-18:
We will look into it ASAP

agrunnet, 2018-04-20:
This is a duplicate thread of this. Will answer other thread. 
https://github.com/IntelRealSense/librealsense/issues/1589

closed 2018-05-14 by RealSense-Customer-Engineering