realsenseai/librealsense #1565: Assets/RealSenseSDK2.0/Scripts/AlignImages.cs(2,7): error CS0246: The type or namespace name `Intel' could not be found. Are you missing an assembly reference?

wxp0821, 2018-04-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

superaldo, 2018-04-19:
Same problem here, after I get the binaries and put in the correct folder on my unity project, everything looks to work but I can´t compile (build) I get this error :Assets/RealSenseSDK2.0/Scripts/RealSenseDevice.cs(8,7): error CS0246: The type or namespace name `UnityEditor' could not be found.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hi @wxp0821, @superaldo,

Did you follow below instructions to successfully build the Unity app?
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity
For the namespace issue, you can refer to below. Try to add a new script like ‘rsEditor.cs’ and move below code in RealSenseDeviceInspector.cs to this new script.
public class RealSenseDeviceInspectorEditor : Editor
{
…
}
https://answers.unity.com/questions/316805/unityeditor-namespace-not-found.html

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
Hi @wxp0821,

Please try librealsense 2.12.0. This should resolve this issue.

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @wxp0821,

Do you have any update?

RealSense-Customer-Engineering, 2018-07-08:
[Realsense Customer Engineering Team Comment]
Hi @wxp0821,

I will close this. If you have any further question, you still can re-open or create another ticket.

closed 2018-07-08 by RealSense-Customer-Engineering