realsenseai/librealsense #1568: extrinsic calibration d415 realsense camera help needed

martijnkusters, 2018-04-18:
I have some issues with doing the calibration of the extrinsic parameter of the d415 camera calibration.

I'm trying two approaches:

1) Using the custom calibration document: 
When reading the custom calibration document, I could not find the customcalibrationsample.zip for download. Can somebody help me where to find these filese.

2) Put a cube on a table and Save the pointcloud to disk and use our inhouse made code in matlab to do the calibration. We used these code before with Kinect v2 camera and that worked.

This is how the program captures the pointcloud.
[image]

When I save the pointcloud, and looked at the data.
I noticed that 3 planes are not completely orthogonal from the pointcloud.
[image]
When I try to do the calibration, i got this results, without demanding the planes to be orthogonal.
[image]
This looks not so bad, but still the raw data from start point was not so good.

Does somebody know how to solve the problem that the planes are not completely orthogonal and to improve the raw data?

Best Regards,
Martijn Kusters

JasonChang13, 2018-04-25:
hi @martijnkusters,

for the sample you are looking for, can you please confirm if examples in DynamicCalibrationAPI-2.4.0.0-Setup.exe are what you are looking for?
download page:
https://downloadcenter.intel.com/download/27415/?v=t

Regards,
Jason
Intel RSG Customer Platform Engineering

martijnkusters, 2018-04-25:
Hi Jason,

I think you are right. But the checkerboard pdf is not included.
In the meantime I captured the depth in high resolution and there the calibration looks much better.

[image]

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
hi @martijnkusters,
please find the calibration chart at link below:
https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html
FYR.

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any follow question.

closed 2018-06-11 by RealSense-Customer-Engineering