realsenseai/librealsense #1569: Given stream configuration DISPARITY16 is not supportet by the device

moecurlyhair, 2018-04-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   { Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    | | 
| SDK Version                            |  { SDK 2.0 }                          | 

### Issue Description

Hello,

i want to get the disparity by using the Intel Realsense D435 with ROS. Therefore i changed the _format[DEPTH] = RS2_FORMAT_Z16 to _format[DEPTH] = RS2_FORMAT_DISPARITY16 at the base_realsense_node. 

When i launch the rs_camera node i get the WARNING: 
[ WARN] [1524061042.124979319]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Format: DISPARITY16, Width: 640, Height: 480, FPS: 30

Is there any possibility to solve this problem or get the disparity?

Tank you!

RealSense-Customer-Engineering, 2018-04-26:
[Realsense Customer Engineering Team Comment]

@moecurlyhair
The Disparity16 format is not used in D400 FW anymore, which is used for old camera FW. Currently we use processing block to exchange depth and disparity data in C++, you can refer to the code of "disparity_transform" inside the disparity-tranform.cpp, which indicate how do we exchange the data in between. Then you will need to implement the function inside ROS wrapper "base_realsense_node.cpp" by referencing how alignFrame(ROS) is being migrated from algin.cpp(C++) processing block.

Since this is the new feature request, I will discuss internally if this can be implemented inside ROS, but be noticed, this shouldn't happen in very short time.

moecurlyhair, 2018-05-02:
@RealSense-Customer-Engineering @dorodnic 
Thanks a lot for your answer!
I will try to implement the function disparity_transform inside the ROS-Wrapper, but I am not sure that I can solve this easily. 

I would appreciate if this issue could be implemented internally by Intel, because I need the disparity to continue my project. 

Thank you and best regards!

closed 2019-01-04 by RealSense-Customer-Engineering