realsenseai/librealsense #1570: Access Camera Intrinsics in Unity Wrapper

gjmsoftware, 2018-04-18:
### Issue Description
My project requires the camera intrinsic information, which changes for each camera (D400). Accessing this information is fairly straight forward in the C# wrapper by calling the "_GetInstrinics()_" method on the "_VideoStreamProfile_" object. However, I'm not sure how to access the "_VideoStreamProfile_"within the current Unity wrapper example. 

I'm hoping someone could provide me insight in how-to successfully call the "_GetInstrinics()_" function in the current Unity wrapper example. 

FYI, the main variables I'm trying to access are: ppx, ppy, fx, and fy. 

----------------------------------------------------------------------------------------------------
### System Information

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | Unity - 5.5.1f1 |
| SDK Version                            |  V2.10.2            |

zivsha, 2018-04-19:
Hi Greg,
If you know how to do it in C# with a device then all you need to do in Unity is get the device in your script.
Please see how it's done in [RealSenseDeviceInspector](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RealSenseDeviceInspector.cs) and follow the same way, which is basically to get the active profile of the RealSenseDevice and take the device from it.

gjmsoftware, 2018-04-20:
Thanks for the guidance @zivsha - it helped and I was able to solve my issue fairly quickly.

closed 2018-04-20 by gjmsoftware

coldcoda, 2020-05-11:
Hello @gjmsoftware ，I also need to get the intrinsics in unity, but I don't know how to find the function "GetInstrinics()". Can youo help me? May be I need a more detailed steps.
Thanks a lot!!!