realsenseai/librealsense #1572: Post Processing implementation for Wrappers.

marklundin, 2018-04-18:
We've been testing out the D435 and have notices quite a noisy feed from the depth. Based on similar issues, it's recommended to use the post processing filters to smooth the results.

Are these filters available via the Unity wrapper?

dorodnic, 2018-05-08:
Will be added to the example scene. All the required components are already exposed in the C# wrapper, so this should not be much different from Align. However, there are couple pressing bugs we are trying to address in the Unity wrapper before we can handle this.

dorodnic, 2018-07-26:
Added to [2.14.1](https://github.com/IntelRealSense/librealsense/tree/development/wrappers/unity#processing-blocks)

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @marklundin,

since 2.14.1 alread released, let me close this ticket here, and if any question, please open new ticket.

closed 2018-08-09 by RealSense-Customer-Engineering