realsenseai/librealsense #1573: Trouble Installing librealsense1 on Ubuntu 14.04

rkeatin3, 2018-04-18:
---------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Ubuntu14.04.1 | 
| Kernel Version (Linux Only)    |  4.4.0-119                                       | 
| Platform                                 | PC                             |
| SDK Version                            |  legacy                          | 

### Issue Description
I made a previous post [here](https://github.com/intel-ros/realsense/issues/365) on the ROS-related repo, but wasn't able to resolve my issues.

Anyway, I am trying to install legacy librealsense. I tried installing via instructions for linux in the 1.9.7 release, but then I get the following error: 

`terminate called after throwing an instance of 'rs::error'
  what():  api version mismatch`

I also tried installing via the deb packages ros-indigo-librealsense and ros-indigo-realsense-camera. I was able to launch the SR300 nodelet, but the depth image flashes weirdly and the pointcloud works intermittently. 

I did install librealsense2 via deb package (though that was troublesome as well), but found that, as expected, there were a number of breaking changes that would require me to rewrite a lot of code already developed for the SR300.

What procedure should I follow? Any help urlwill be greatly appreciated!

rkeatin3, 2018-04-19:
I followed those steps exactly, but when I try to load the `uvcvideo` kernel module I get the following error:
`modprobe: ERROR: could not insert 'uvcvideo': Exec format error`

Was installing llibrealsense2-dkms intended to solve that problem?

Also note that you linked to a tagged release of the `realsense_camera` ROS package but then instructed me to install the package via `apt-get`. Either way, that confusion isn't the main problem at the moment as I can't load the necessary kernel module. 

Lastly, in running through those steps a second time to ensure I had followed them properly, I see that the script for installing `glfw3` installs `cmake` and the following step uninstalls `cmake` in favor of `cmake3`. Does that script need to depend on the older version of `cmake`?

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hi @rkeatin3,

Do you resolve the issue?

rkeatin3, 2018-05-16:
No, this issue did not get resolved. I ended having to port all of my code over to librealsense2 to get around the apparent compatibility issues.

RealSense-Customer-Engineering, 2018-06-07:
[Realsense Customer Engineering Team Comment]
Hi @rkeatin3,

Since this should be the compatibility issue on your build system, I will close this. If you have any concern, you can re-open it.

closed 2018-06-07 by RealSense-Customer-Engineering

wucm1994, 2018-07-19:
Please refer to the solution I provided in [here](https://github.com/intel-ros/realsense/issues/365)

Hope this helps.