realsenseai/librealsense #1574: Build error caused by -mavx2 option

LostCanadian, 2018-04-18:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.2 | 
| Operating System & Version | Ubuntu 16.04 | 
| Kernel Version    | 4.4.77-rockchip-ayufan-136 | 
| Platform                                 | Rock64 |
| SDK Version                            | 2.10.4 | 

With the latest SDK release, I am running into compilation errors due to the new AVX features. I was hoping the new release would fix all the AVX-related issues (see: #1507), but this is an issue at the build configuration level (probably). Is there an additional flag I need to pass to CMake so that it doesn't try to use that flag at all? I assume this issue crops up because I'm on an ARM64-based architecture, with no AVX support.

Here's my cmake command (nothing special here):
`cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false -DCMAKE_BUILD_TYPE=release`

Here's the error when doing a make:
`c++: error: unrecognized command line option '-mavx2'
CMakeFiles/realsense2.dir/build.make:398: recipe for target 'CMakeFiles/realsense2.dir/src/image_avx.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/image_avx.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2`

I'm on version 5.4.1 of g++/gcc.

Note that I was able to compile and stream data from the D415 (albeit with some stability and throughput issues) using the v2.10.0 release.

dorodnic, 2018-04-19:
Hi @LostCanadian 

The "mavx2" option is supposed to be supported by GCC version 4.7 and higher.
Could you please double check the gcc version (`$ gcc --version`)?

In general, RealSense products are not officially validated on ARM architecture. We are happy to respond to user feedback and integrate patches for better support. 

You can choose between rolling-back to 2.10.0 version or modifying `CMakeList.txt` (just delete the `-mavx2` flag).

LostCanadian, 2018-04-19:
Hi dorodnic,

As mentioned, the gcc version is 5.4.1 (according to gcc --version). I suspect that this flag is not supported in gcc for ARM platforms. 

Removing this line from CMakeLists.txt does the trick:
`set_source_files_properties(src/image_avx.cpp PROPERTIES COMPILE_FLAGS -mavx2)`

If there is some way to omit that line conditionally based on the architecture, that should be all that is needed to fix the build for ARM platforms.

dorodnic, 2018-04-19:
Glad it worked! 
We will try to add automatic detection in the upcoming sprint.

mfassler, 2018-04-20:
Just wanted to say "me too".  I'm building on ARM (the Jetson TX2).  ARM doesn't support SSE or AVX.  Removing this line in CMakeLists.txt worked for me.

akatrevorjay, 2018-04-30:
Lol, I ran into this last week too, just hadn't gotten to put it in a PR yet. Quick Q from other TX2 owners -- Have you been able to sustain enough USB 3 bandwidth to use a D4xx without it crashing hard until a reboot is performed?

mfassler, 2018-05-01:
I have been able to sustain a raw data capture at about 220 MiB/s for a
couple of minutes with no problems.  This was an Intel RealSense D435 ->
usb3 -> tx2 on an Auvidea J120 board -> M.2 (pci) SSD

closed 2018-05-08 by dorodnic

JeffMII, 2021-11-04:
So we're in 2021 now. Has this process not been fixed for ARM architectures? I'm sure you know that Raspberry Pi is ARM and quite popular these days. How would I go about deleting these problematic lines from CMakeLists.txt? I'm trying to `pip install spacy`.