realsenseai/librealsense #1576: How can I use the unity wrapper with the MacOS library?

krstns, 2018-04-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   MacOS  | 
| Platform                                 | PC  |
| SDK Version                            |  2.2.10.4 }                          | 

### Issue Description

I want to use the unity wrapper on MacOS but it comes only with the dll's. Is it possible to make it work with the librealsense2.2.10.4.dylib?
Or is there something else involved? I can see two dll's are required when running on windows.

Thanks

eccoscreen, 2018-04-25:
Has anyone figured out how to do this? I tried the above comment but I'm not sure if I did it right.

krstns, 2018-04-25:
@eccoscreen I can build the library on macos using the guide from github, but I can't get the wrapper to build. I am no fan of mono and maybe that's why. I will give it a go in the coming days, but if you will figure it out before, please let me know how i can get that done.

krstns, 2018-05-18:
Actually, it was quite simple. Just load the sources in Visual Studio for Mac 2017, select the wrapper and build.

closed 2018-05-18 by krstns

krstns, 2018-05-19:
So, what I did was:
follow the guide of getting the librealsense on macos as described in here: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

Then I have installed the Visual Studio 2017 for Mac (BTW, well done Microsoft, you have delayed the decission of moving my intellij license to my windows installation by 30minutes... which is 30 minutes longer than before ).

Then I have loaded the csharp wrapper solution inside the VS and simply hit build. Mind you, you might want to change the type from Debug to Release.

It will give you a DLL file, with the same name as the one attached to the Realsense unity package, I have changed the extension to .dylib.

Now copy the dylibs created from the librealsense project and the Intel.Realsense.dylib you have just built to the Plugins directory where you have the original original unity asset's dlls.

Then in Unity, it will give you quite some errors. You have to select the original DLLs and in inspector mark them as macos only. In case of the original DLLs, mark them as windows only.

From my memory, this was enough to get it going and working. Remember to attach your camera to the correct port (best check with the Realsense viewer, it has the indication whether it is USB 2.1 or 3 or 3.1). In my case (old 2013 macbook pro) i had to use the right usb and Realsense viewer showed it as 3.1 (which i don't have ;) ). Anyway, then I was able to select the resolutions I wanted. Once you are done there, kill realsense viewer and apply the same resolution and framerate configuration there. 
After this, it started to work for me.

If you have any questions, I will be happy to answer, it was not detailed because I am away from my laptop and on windows at the moment.

eccoscreen, 2018-05-23:
@krstns  Yeah I'm still getting a bunch of 'Intel' could not be found errors. What are the names of the files you've added and in which folders did you add them to?

krstns, 2018-05-23:
@eccoscreen here's a little screenshot with all the details you asked for.

[image]

cansik, 2018-11-05:
I also created an example unity project for MacOS and the D435. Running smooth now, but rosbag playback and some other things are not working:

https://github.com/cansik/realsense-unity-mac

CharStiles, 2019-03-07:
@krstns I was wondering how you compiled the csharp wrapper solution on mac, I tried by downloading visual studio for mac and following the instructions to generate the sln, but cmake wont let me choose my visual studio as the generator. 
At the end of the day I am trying to just get the realsense working in unity on a mac, unfortunately the example @cansik made is crashing unity for me.

thanks so much for any help, i am very green to all of this, so any pointers are appreciated!