realsenseai/librealsense #1577: RealSense viewer - Save settings fails

kirsitn, 2018-04-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415                                                      | 
| Firmware Version                   |  05.08.15.00                                            | 
| Operating System & Version |  Linux (Ubuntu 16)                                 | 
| Kernel Version (Linux Only)    |  4.4.38                                                    | 
| Platform                                 |  NVIDIA Jetson TX2                                |
| SDK Version                            |   2.9.1                                                    | 

### Issue Description

I am trying to save my current settings from the RealSense viewer, but it fails nearly every time with the error message "The profile is unsupported!".  This happens even if I try to save a standard preset such as "high density". The only time I am able to save settings is if I do it right after opening the viewer and with settings chosen from the drop-down menu of presets.  Once I have loaded settings from file, I can no longer save any settings (neither standard settings nor the loaded settings). The camera runs normally with the settings I wish to save, so the error message does not seem to have any connection with the settings themselves.

dorodnic, 2018-04-19:
Hi @kirsitn 
We will try to reproduce this issue and look into it. Could you please provide more information? Perhaps the camera is identified as USB2? Did you try the newer SDK versions? Did you see this problem outside the Jetson (perhaps on the desktop PC)?

kirsitn, 2018-04-19:
The camera is not identified as usb2 (that happens as well from time to time, but then I'm not able to load the settings at all)

I have not tested other SDK versions.

I will see if I can find a pc where I can test it tomorrow.

---- Sergey Dorodnicov wrote ----

Hi @kirsitn<https://github.com/kirsitn>
We will try to reproduce this issue and look into it. Could you please provide more information? Perhaps the camera is identified as USB2? Did you try the newer SDK versions? Did you see this problem outside the Jetson (perhaps on the desktop PC)?

kirsitn, 2018-04-20:
I have now tested on a PC with Linux and with Windows, and in both cases the result is the same for version 2.9.1 and the most recent version (2.10.x): I can load settings, but not save them again (the settings file I try to save to is created, but not written to). 

Version 2.9.0, however, allows me to both load and save settings (only tested on Windows). The menu for saving looks different in that version, so it seems like the bug may have been introduced when the menu was changed.

RealSense-Customer-Engineering, 2018-05-30:
[Realsense Customer Engineering Team Comment]
hi @kirsitn,

I can reproduce this with below steps on windows, please let me know if it's not the same with yours.
1. start realsense viewer, change preset to "high density" and save preset to test.json.
2. load test.json, do nothing, and save to test.json again, this will report "The profile is unsupported"

kirsitn, 2018-05-30:
Not entirely the same. I get the "profile is unsupported" message already when I try to save the chosen preset in your step 1.

RealSense-Customer-Engineering, 2018-06-01:
[Realsense Customer Engineering Team Comment]
hi @kirsitn,

thanks for confirm, do you mean you got the error message even the test.json not even existed?
is this happen on both windows PC and Linux?
I reproduced on windows so I am wondering if this case is Linux only.

kirsitn, 2018-06-04:
It happens both on Windows and Linux and also when saving the settings to a new file. The file is created, but not written to (0 bytes).

AntoineWefit, 2018-06-05:
Hi @kirsitn 
You might want to check if the "Color" option is ticked in "RGB Camera"; if not, tick it and try to save again.
That did the trick for me :)
[image]
Keep us updated

kirsitn, 2018-06-05:
That actually seems to work! Thank you! (It seems that color is automatically unchecked when I load some of the settings, and I didn't expect the choice of streams to have any impact on saving of settings.)

AntoineWefit, 2018-06-06:
Yes (in my case) the option is disabled every time I load one of my own ".json". It doesn't prevent you from streaming though, but give the error you mentioned if you try to save the configuration. 
I don't really know if we missed some settings to save this parameter or if it's some kind of bug @dorodnic ?
Also, if you have good knowledge of the parameters labels, you can bypass the error by writing manually your parameters values in the ".json" files without using the "Save current settings to file" of Intel RealSense Viewer, but it's not really convenient...

Anyway, if you think that "solved" your problem, you might consider closing this issue.

RealSense-Customer-Engineering, 2018-06-06:
[Realsense Customer Engineering Team Comment]
hi @AntoineWefit,

thanks for the workaround, and we have this listed as known issue in v2.12.0 release, this is on the list to be fixed.

kirsitn, 2018-06-06:
"Also, if you have good knowledge of the parameters labels, you can bypass the error by writing manually your parameters values in the ".json" files without using the "Save current settings to file" of Intel RealSense Viewer, but it's not really convenient..."

That's more or less what I did, but since it takes a while to find all the correct parameter names, I only changed the most important ones. Now I will hopefully be able to save also the less important ones. automatically :-)

closed 2018-06-06 by kirsitn

tispratik, 2019-05-24:
The work around does not seem to work for me, I tried it multiple times.

AntoineWefit, 2019-05-27:
Hi @tispratik 
Which version of the Viewer are you using ?
Are you having the exact same problem than the original one ? If not, you might want to open a new issue.

tispratik, 2019-05-27:
I am using the latest available version for download as of today. Thanks!

AntoineWefit, 2019-05-27:
> The post processing filter details for depth are not getting saved.

It's not a bug, those parameters can't be saved in the ".json".

You can open your parameters files saved with the Viewer with a text editor to know exactly which settings are actually saved, their names are pretty much self-explanatory :) 

Here is an example : 
```
{
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.436523",
    "aux-param-colorcorrection10": "-0.553711",
    "aux-param-colorcorrection11": "-0.553711",
    "aux-param-colorcorrection12": "0.0458984",
    "aux-param-colorcorrection2": "0.540039",
    "aux-param-colorcorrection3": "0.540039",
    "aux-param-colorcorrection4": "0.208008",
    "aux-param-colorcorrection5": "-0.332031",
    "aux-param-colorcorrection6": "-0.212891",
    "aux-param-colorcorrection7": "-0.212891",
    "aux-param-colorcorrection8": "0.68457",
    "aux-param-colorcorrection9": "0.930664",
    "aux-param-depthclampmax": "7000",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "100",
    "controls-autoexposure-auto": "False",
    "controls-autoexposure-manual": "30000",
    "controls-color-autoexposure-auto": "False",
    "controls-color-autoexposure-manual": "312",
    "controls-color-backlight-compensation": "1",
    "controls-color-brightness": "10",
    "controls-color-contrast": "46",
    "controls-color-gain": "46",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "70",
    "controls-color-sharpness": "74",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-depth-white-balance-auto": "False",
    "controls-laserpower": "180",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "400",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "7000",
    "param-depthclampmin": "0",
    "param-depthunits": "100",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "100",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "166",
    "param-maxscorethreshb": "565",
    "param-medianthreshold": "453",
    "param-minscorethresha": "0",
    "param-neighborthresh": "0",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "17",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "1",
    "param-scanlinep1onediscon": "261",
    "param-scanlinep1twodiscon": "244",
    "param-scanlinep2": "511",
    "param-scanlinep2onediscon": "258",
    "param-scanlinep2twodiscon": "246",
    "param-secondpeakdelta": "1023",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "100",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-ir-format": "Y8",
    "stream-width": "1280"
}
```

tispratik, 2019-05-27:
Thanks for the information @AntoineWefit, I am looking for decimation filter, threshold filter min, max. I don't seem to find them in the json. Is setting through the api using https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb the only option?

AntoineWefit, 2019-05-28:
> I am looking for decimation filter, threshold filter min, max. I don't seem to find them in the json.

Those are post-processing filters, as I said, **you can't save them** in the configuration file.

> Is setting through the api using https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb the only option?

You can set them manually within the Viewer (but they won't be saved), or write a bit of code with the help of the SDK.
Here is [the page](https://github.com/IntelRealSense/librealsense/tree/9bdbfd98440b6516f8c864fbdfd6a9778b2c5715/src/proc) with the post-processing filters.
You should also take a look at [this example](https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing) to begin with.

tispratik, 2019-05-28:
Thanks Antoine. The viewer seems to persist the IR emitter on/off setting on the camera. It was a bit surprising that a "view"er is changing some values on the camera. That led me to believe that other settings like the post processing filters may also be saved.

Thanks for the clarification and links to the docs and examples.