realsenseai/librealsense #1580: Discovery of not supported devices via query_devices()

fwindolf, 2018-04-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0-38                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2                          | 

### Issue Description

I am experimenting with two cameras plugged in at the same time, one Asus Xtion2 and one D435.
```
// Discover devices
rs2::context ctx;
auto deviceList = ctx.query_devices();

for (auto device : deviceList)
{
    // Find out device name
    std::string name = "Unknown Device";
    if (device.supports(RS2_CAMERA_INFO_NAME))
        name = device.get_info(RS2_CAMERA_INFO_NAME);

    std::cout << "Discovered device: " << name << std::endl;
}
```
Next to the discovered RealSense, the library first discovers the RealSense as "Intel RealSense 435" and then seems to discover the Xtion2 as "Platform Camera" twice. When reading their name to weed out the devices not accessible with librealsense, each device access taking about 5-10 seconds.

dorodnic, 2018-04-19:
Hi @fwindolf 
You can use librealsense to stream data from regular web-cams. 
We might add a flag to `query_devices` to control this behavior. In the meantime you can disable unwanted behavior by commenting out [context.cpp:336-337](https://github.com/IntelRealSense/librealsense/blob/master/src/context.cpp#L336-L337)

fwindolf, 2018-04-25:
Thanks @dorodnic for the quick reply.
I commented out the said lines and it works like a charm!

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
hi @fwindolf,

do you need more support on this topic?
thanks.

RealSense-Customer-Engineering, 2018-06-01:
[Realsense Customer Engineering Team Comment]
let me close this issue here, please feel free to open new ticket if any question.

closed 2018-06-01 by RealSense-Customer-Engineering