realsenseai/librealsense #1585: Interlacing, artifacts and double objects issues. Distorted color frame and depth values.

SimonLit, 2018-04-20:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Windows 10 |  
| Platform                                 | Qt5.10  |
| SDK Version                           | 2.10.1 | 

For the last two days I have been having an issue with the colour frames that are coming out of the RealSense D415 camera. The frame looks like the image below. The way it looks happens 9 out of 10 times that I run the application. Sometimes it is fine however. The colorized depth frame looks fine, without distortions, tough the depth value of pixels is around 60 meters which is clearly not possible with this camera.  

[image]

I am using Qt5.10 as framework for the application. The image is displayed as a QPixmap but before that is being converted from RS2_FRAME to QImage with Opencv3. The code for setting the streams and converting the images is below. 

Enabling streams
`    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 0, 640, 480, RS2_FORMAT_RGB8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 0, 640, 480, RS2_FORMAT_ANY, 30);
    cfg.enable_stream(RS2_STREAM_INFRARED, 0, 640, 480, RS2_FORMAT_RGB8, 30);
`
Converting colour frame to opencv Mat and saving it as QImage.
`   //Color frame
        cv::Mat _color(cv::Size(ret.width, ret.height), CV_8UC3, (void*)data.get_color_frame().get_data(), cv::Mat::AUTO_STEP);
        ret.colorFrame = QImage(_color.data, _color.cols, _color.rows, _color.step, QImage::Format_RGB888);
        //Depth frame
        cv::Mat _depth(cv::Size(ret.width,ret.height), CV_8UC3, (void*)depth_frame_colorized.get_data(), cv::Mat::AUTO_STEP);
        ret.depthFrame = QImage(_depth.data, _depth.cols, _depth.rows, _depth.step, QImage::Format_RGB888);
        //Infrared frame
        cv::Mat _leftInfrared(cv::Size(ret.width, ret.height), CV_8UC3, (void*)data.first(RS2_STREAM_INFRARED).get_data());
        ret.leftInfraredFrame = QImage(_leftInfrared.data, _leftInfrared.cols, _leftInfrared.rows, _leftInfrared.step, QImage::Format_RGB888);
`
What have I tried so far:

- Display image in a different way, QPainter, Graphicsview, QLabel
- Used wait_for_frames() aswell poll_for_frames()
- Made sure the data was not accessed by multiple threads at the same time (mutex)
- Unplugged camera between runs of application
- Reset to factory
- Enable settings like advance mode through Depth tool instead of own code

With what do i need help? 

- Determining if it is a known problem
- Where may lay the fault looking at the distortions in the image
- Finding solutions, pointers to where to look next
- New tests to find the problem

In advance many thanks.
Simon

vladsterz, 2018-04-20:
`cfg.enable_stream(RS2_STREAM_INFRARED, 0, 640, 480, RS2_FORMAT_RGB8, 30);` Are you sure this is correct?

SimonLit, 2018-04-20:
Not 100% but I shall play around with it to see if it has a effect. Any clue to what the correct RS2_FORMAT would be for infrared?

vladsterz, 2018-04-20:
I am pretty sure that this is wrong. I am not sure what is should be tho.  Either RS2_FORMAT_Y8 or RS2_FORMAT_Y16. Firstly try RS2_FORMAT_Y8

dorodnic, 2018-04-22:
Hi @SimonLit 
Possible one of the opencv examples can be useful as reference: 
[rs-latency-tool.cpp](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/latency-tool/rs-latency-tool.cpp)

dorodnic, 2018-04-23:
Hi @SimonLit 
Did you managed to make it work?

SimonLit, 2018-04-23:
Hi @dorodnic ,
For the moment I haven't solved the problem. I am pretty sure the transfer of the data between threads is going wrong. So I am going to close this issue for now. If I find a proper solution later I will post it here for the community. 

Thanks for the effort.

closed 2018-04-23 by SimonLit