realsenseai/librealsense #1586: C Examples freeze on MacOS

edwinRNDR, 2018-04-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   MacOS 10.11 and 10.13  | 
| SDK Version                            | 2.2.10.4 / 930b39cbfe381165e8c56df9299d9cad97ab1acb |                         

All of the C examples wait indefinitely for data after `rs2_pipeline_start_with_config`. The other examples like `rs-measure` work fine.

Additionally. I am working on Java wrappers for librealsense and I am running into segfaults on OSX 10.11 and a similar (same?) freeze on MacOS 10.13. I have a stack trace of that here: https://gist.github.com/edwinRNDR/88adee566991e112f1a23087987ac258

drhalftone, 2018-04-25:
I have the exact same issue running RS 2.10.4

drhalftone, 2018-04-26:
I can now confirm the issue using the latest firmware 5.9.9.2.

Gnimuc, 2018-04-30:
same here. My current workaround is add `sleep(1);` before `rs2_pipeline_start_with_config`.

drhalftone, 2018-04-30:
Adding that sleep worked for me.

Gnimuc, 2018-05-03:
`sleep` won't work in the record-playback example where multiple pipelines are needed to be created and deleted.

drhalftone, 2018-05-03:
Funniest thing, I just ran a version of my code that puts the RealSense inside a QTCPServer object so that I can stream video over a network. And a 1.0 second sleep didn't do it.  I had to extend it to 2.0 seconds.  I don't know if this is meaningful but it is a symptom.  Maybe it can help out Intel in figuring out what's wrong.

drhalftone, 2018-05-07:
I can't help but wonder, given the lack of responses from Intel, if they consider this a problem with the library or with us?

dorodnic, 2018-05-08:
Hi @drhalftone @edwinRNDR @Gnimuc 
I apologize for long response time. This is definitely our bug that needs to be fixed, but the team wasn't able to get to it just yet.

Gnimuc, 2018-05-15:
I just found the low-level API `rs2_open` has this problem too. I need to pause between choosing a profile and starting the sensor.
```
profile = choose_a_streaming_profile(sensor)

sleep(1)

rs2_open(sensor, profile, err)
checkerror(err)
```
This info might be helpful for narrowing down the bug.

RealSense-Customer-Engineering, 2018-05-17:
[Realsense Customer Engineering Team Comment]
we are still working on this issue, and thanks for your information.

jcelerier, 2018-05-22:
I could fix the hang by doing the following change : https://github.com/IntelRealSense/librealsense/pull/1749

stumacd, 2018-06-14:
Hi,
I had the same problem and a sleep didn't solve it for me. Realsense-viewer works for me in a limited capacity (the resolution is fixed at 480 x 270, the device has mounted as USB2 and there is no RGB stream - similar issues as other people in the thread). So in the example I changed the default Height and Width to 480 and 270 respectively and the example works.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
hi @edwinRNDR, @drhalftone, @Gnimuc,

Proper fix is now in the development branch, will be released soon

dorodnic, 2018-07-22:
Should be resolved now in 2.14.0

dorodnic, 2018-07-26:
I assume this is resolved, if for some reason we are still missing something, please open a follow-up issue

closed 2018-07-26 by dorodnic