realsenseai/librealsense #1587: Issue serializing json in adv mode: Value not found in map! value=8

wingillis, 2018-04-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  MacOS 10.13.4 | 
| SDK Version                            |  2.10.3    | 

### Issue Description
When running the example python script `python-rs400-advanced-mode-example.py` to get the advanced mode parameters, I receive normal output for each `get` call for the parameters, but once the line running `serialize_json()` (line 72) is run, I get an exception: `Value not found in map! value=8`. I don't know if this is an issue with the python wrapper or to c++ serialization algorithm.

At first I thought it was a mismatch between the firmware and the SDK, so I updated the firmware. Serializing worked after the firmware update, but after I tried to set a parameter and serialize, I started receiving the same `Value not found in map! value=8` error.

dorodnic, 2018-04-22:
Hi @wingillis 
We will look into it.

wingillis, 2018-04-22:
Thanks! I should note that I also see this error when I run a similar program written in c++, so I don't think the issue is constrained to the python wrapper, but maybe the serializer, or wherever the parameters are saved on the realsense itself.

RealSense-Customer-Engineering, 2018-06-13:
[Realsense Customer Engineering Team Comment]
hi @wingillis
finally I finish my setup and  I can reproduce this issue and confirmed v2.12.0 still having this issue.
at the same time, python-tutorial-1-depth.py is working good.
will continue investigate on this.

wingillis, 2018-06-13:
Great thanks for checking in! Yeah I still receive this error as well, even
with the new SDK and firmware updates. I am able to load json parameters
though.

neubotech, 2018-07-12:
I have the same problem.  I'm trying to load the advanced mode so that I can use different settings of D435. I'm wondering if anyone has solved this issue.  Seems like a c++ level issue.

dorodnic, 2018-11-05:
Should be resolved in latest release

PR #2490 (2018-10-11): Libuvc fixes
Added a function that converts UVC auto-exposure mode values to RS compatible 0/1 values. 
The auto exposure modes are introduced in the [UVC specs](http://www.cajunbot.com/wiki/images/8/85/USB_Video_Class_1.1.pdf#page=94). D0 (manual exposure time) is converted to 0, D3 (auto exposure time) is converted to 1.
Previously D3 (as byte array) was converted to integer with value 8, which caused issues as in librealsense auto exposure value is expected to be either 0 or 1.
Fixes: #1587

Additionally, fixed encoding issues in python advanced mode example.
closed 2018-11-08 by dorodnic