realsenseai/librealsense #1589: Intel Realsense camera and labview

lonely-users, 2018-04-20:
Hello,
      I want to use Labview to develop Realsense d435 in Windows 10, I found a set of instanceshttps://github.com/IntelRealSense/librealsense/tree/development/wrappers/labview，but there is no way to use it after download, subVI is missing and the .dll file cannot be imported(missing corresponding header file).
     Who can give me some suggestions or tips?
     Thank you!
[image]
[image]
[image]

agrunnet, 2018-04-20:
Something appears to be blocking and cleansing your downloads of important files. I just retried and it is all there. 
We can try to look at password protecting it and posting a new version here. Maybe that will allow you to download.

agrunnet, 2018-04-20:
I think we can solve this by assuming the DLL's got removed, so you need to generate new DLL's. 

In this document 
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/labview
it says
"The realsense2.dll that is included in this package is 64bit for Windows 10. Other versions (ex: 32bit) could be compiled by following the instructions on GitHub. Just rename them realsense2.dll and replace the file in the location used by the LabVIEW demos."

Please make sure to compile the LibRS to the 32bit or 64bit version you have. 
Finally, if you still have an issue, drill down to the SUBVI and the "Call library function node" and check that the path points to the proper DLL. 
[image]

lonely-users, 2018-04-21:
@agrunnet However, I have problems importing .dll files into LabVIEW. There is no way to import them into labview correctly, so there may not be a way to check this. Can you tell me the detailed steps? I am new to this field

lonely-users, 2018-04-21:
@agrunnet My function pointer is normal. It may be because I didn't know which header file to use so I didn't import it correctly. How should this be solved?

agrunnet, 2018-04-21:
No header is needed. The dll can basically be anywhere. All the important subvi’s have the call function node at their core. This is how you wrap a dll with Labview. You just need to double click on it and point open the path and find the dll file. 

Tell me more about your system and I can make sure we make this easier for you. Windows 10? 64bit? Labview 2017?

lonely-users, 2018-04-21:
Yes, I am using Windows10 64 bit and Labview 2017

lonely-users, 2018-04-21:
@agrunnet How to solve the loss of that SUBVI?  there are missing a lot of VI in the Labview library, the problem described in my issue picture inside.

agrunnet, 2018-04-21:
I will post version that is zipped with password soon.

agrunnet, 2018-04-21:
Okay, try this one. 
[RealSense SDK2.10.4 for LabVIEW.zip](https://github.com/IntelRealSense/librealsense/files/1934912/RealSense.SDK2.10.4.for.LabVIEW.zip)
Use password "intel123"

agrunnet, 2018-04-21:
Here is newer version with an added example for pointcloud with post-processing
[RealSense SDK2.10.4 for LabVIEW.zip](https://github.com/IntelRealSense/librealsense/files/1935033/RealSense.SDK2.10.4.for.LabVIEW.zip)
password:intel123

lonely-users, 2018-04-22:
@agrunnet Thank you very much, I try it.

lonely-users, 2018-04-22:
@agrunnet  Sorry, I forgot that my labview is 32-bit. How does this need to be solved?

agrunnet, 2018-04-22:
Since Labview is just a wrapper on the librealsense DLL, you basically just compile librealsense as 32bit and then copy and replace the DLL in the Labview examples. 
So since this may take some time unless you are all set up to do it, I did it for you here. 
[realsense2_32bit.zip](https://github.com/IntelRealSense/librealsense/files/1935806/realsense2_32bit.zip)
password is intel123

lonely-users, 2018-04-23:
@agrunnet Where do you have realsense2ex.dll? RealSense_HelloWorld_PointCloud Plus2.vi cannot be opened for the 32-bit version of realsense2ex.dll missing.
thank you!

agrunnet, 2018-04-23:
I don’t have that for 32bit yet.

agrunnet, 2018-05-14:
I am posting here a new version of the LabVIEW wrapper based on LibRS 2.11.0.
- It includes dlls for both the 32bit and 64bit version (default) of the librealsense2.dll. 
- It includes the new post-processing that is in the new LibRS, so no need for the extra Librealsense2ex.dll. Look at the example called RealSense_HellowWorld_PointCloud Plus4.vi

[RealSense SDK2.11.0 for LabVIEW.zip](https://github.com/IntelRealSense/librealsense/files/2001607/RealSense.SDK2.11.0.for.LabVIEW.zip)
Password is intel123

lonely-users, 2018-05-15:
@agrunnet Thank you very much for your continuous help and support. I will debug it immediately.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@lonely-users
If the solution provided by Agrunnet solves your problem, please help to close the issue, thank you.

closed 2018-05-23 by RealSense-Customer-Engineering