realsenseai/librealsense #1591: Holes Filling doesn't works correctly with RealSense SDK v2.10.4

UnaNancyOwen, 2018-04-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| SDK Version                            |  2.10.4                                                    | 

### Issue Description
This is a bug report.
Holes filling of <code>rs2::spatial_filter</code> doesn't works correctly  with RealSense SDK v2.10.4.
The holes doesn't filled even if set option to filling all holes.
That filter worked correctly with RealSense SDK v2.10.3 (or older versions).

### How To Reproduce This Problem
Set the option to filling all holes as follows. But, filter output is not filled all holes.
This problem can also be confirmed by setting the [Stereo Module]>[Post-Processing]>[Spatial Filter]>[Holes filling] option to "Unlimited" in RealSense Viewer v2.10.4.
```cpp
rs2::spatial_filter spatial_filter;
if( spatial_filter.supports( rs2_option::RS2_OPTION_HOLES_FILL ) ){
    rs2::option_range option_range = spatial_filter.get_option_range( rs2_option::RS2_OPTION_HOLES_FILL );
    spatial_filter.set_option( rs2_option::RS2_OPTION_HOLES_FILL, option_range.max );
}
```

ev-mp, 2018-04-23:
Hello @UnaNancyOwen , thanks for reporting this.
We indeed unintentionally broke this functionality in the latest v2.10.4 release.

We're working to rectify this and will update this thread accordingly.
As a stop-gap measure you can disable the depth->disparity converter to get the holes-filling option working.

UnaNancyOwen, 2018-04-23:
@ev-mp Thank you for working on fix.
By the way, I have a question about filtering.
What is the meaning of converting to disparity map before filtering?
(It seems to be doing in some samples. But, I don't understand that meaning.)

daversintel, 2018-04-23:
DS cameras use stereo algorithms to get the disparity map, and the depth map is from the disparity map. It's always better to run the post processing algorithms on disparity map.

agrunnet, 2018-04-23:
The main reason is that the depth error scales quadratically with distance. So you could in stead have a spatial filter that is distance dependent -  that filters less aggressively farther away, but it is better to do it all in the disparity domain.

UnaNancyOwen, 2018-04-29:
I tried PR #1603. It works correctly.
Thank you for fixing! 👍

closed 2018-04-29 by UnaNancyOwen