realsenseai/librealsense #1592: Mac Node test-devicelist.js hangs

cxbrooks, 2018-04-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   MacOS  High Sierra 10.13.4 (17E199)| 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | Mac  |
| SDK Version                            |  GitHub                         | 

### Issue Description
The node interface under MacOS High Sierra with node 9.11.1 seems non-operational.

For example, the test-devicelist.js test hangs.

I downloaded, configured and built with

```
    git clone https://github.com/IntelRealSense/librealsense
    cd librealsense
    mkdir build && cd build
    cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G "Unix Makefiles"
    make
    make install
```
As a check, I confirmed that vrious programs like /usr/local/bin/rs-capture work for me.

To build and install the node interface, in librealsense/wrappers/nodejs:

```
    npm install -g jsdoc
    npm install -g node-gyp 
    node install
```

node install failed with:
```
  SOLINK_MODULE(target) Release/node_librealsense.node
clang: error: no such file or directory: '/Users/cxh/src/ptII11.0.devel/librealsense/wrappers/nodejs/../../build/Release/librealsen
```

This is probably because I'm using Unix Makefiles and not XCode, so I did:

```
bash-3.2$ mkdir ../../build/Release
bash-3.2$ (cd ../../build/Release; ln -s ../libreal* .)
bash-3.2$ ls ../../build/Release/
librealsense2.2.10.4.dylib      librealsense2.dylib
librealsense2.2.dylib
bash-3.2$
```

`node install` worked, but running the mocha tests hung.

I then upgraded from node 7.x to 9.x:

```
brew install node
export PATH=/usr/local/Cellar/node/9.11.1/bin:/usr/local/Cellar/node/9.11.1//libexec/bin:${PATH}
cd librealsense/wrappers/nodejs/
rm -rf node_modules
mkdir node_modules
npm install -g jsdoc
npm i npm
npm i npm
npm install -g jsdoc
npm install -g node-gyp
npm install mocha
```

Here are my node and npm binaries
```
bash-3.2$ which node
/usr/local/Cellar/node/9.11.1/bin/node
bash-3.2$ node --version
v9.11.1
bash-3.2$ which npm
/usr/local/Cellar/node/9.11.1//libexec/bin/npm
bash-3.2$ npm --version
5.6.0
```

In librealsense/wrappers/nodejs/test/, running

```
../node_modules/.bin/mocha test-devicelist.js 
```

hung, so I unplugged the camera, replugged it in and verified that  /usr/local/bin/rs-enumerate-devices  worked.  Then, running mocha got further, but still hung, so I added console.log() statements to librealsense/wrappers/nodejs/test/test-devicelist.js:

```
  it('Testing member size', () => {
    console.log('test-devicelist.js: testing member size start');
    assert.equal(typeof devl.size, 'number');
    console.log('test-devicelist.js: testing member size end');
  });

  it('Testing member back', () => {
    console.log('test-devicelist.js: testing member back start');
    let dev = devl.devices;
    assert(dev.length > 0); // Device must be connected                                        
    let device = devl.getDevice(dev.length -1);
    let SN = device.getCameraInfo().serialNumber;
    assert.doesNotThrow(() => {
      if (devl.size > 0) {
        let res = devl.back;
        assert(devl.back instanceof rs2.Device);
        assert.equal(typeof res, 'object');
        assert.equal(res.getCameraInfo().serialNumber, SN);
      }
    });
    console.log('test-devicelist.js: testing member back end');
  });
```

Now, I get:

```
bash-3.2$ ../node_modules/.bin/mocha test-devicelist.js 
test-devicelist.js: before testing member size

  DeviceList test
test-devicelist.js: beforeEach() start
test-devicelist.js: beforeEach() end
test-devicelist.js: testing constructor
    ✓ Testing constructor
test-devicelist.js: beforeEach() start
test-devicelist.js: beforeEach() end
    ✓ Testing member devices
    1) "after each" hook for "Testing member devices"

  2 passing (2s)
  1 failing

  1) DeviceList test
       "after each" hook for "Testing member devices":
     Error: Timeout of 2000ms exceeded. For async tests and hooks, ensure "done()" is called; \
if returning a Promise, ensure it resolves. (/Users/cxh/src/librealsense/wrappers/nodejs/test/\
test-devicelist.js)

bash-3.2$ 
```

Has anyone had any success recently with the Librealsense Node interface under macOS?

tingshao, 2018-04-27:
Did you tried the examples? please follow  the steps:
1. enter wrappers/nodejs/examples
2. `npm install`
3. `node nodejs-capture.js` or some other examples

tingshao, 2018-04-28:
We tested the examples under 'wrappers/nodejs/examples' directory on mac 10.12.6 not from source but using the npm package `node-librealsense` which was a previous released version. I will try the latest source when I get a mac device.

tingshao, 2018-04-28:
I performed a build from source verification on my Mac:
MacOS high sierra 10.13.4
node 9.11.1
Camera: D435

The example nodejs-capture.js also worked. 
I also found some hung issue when I run mocha test. Those tests work normally on Ubuntu 16.04. 
It seems there maybe something wrong with the objects life cycle and the timeout made it worth.

cxbrooks, 2018-05-01:
Whoo Hoo!  Upgrading the firmware from 05.08.15.00 to 05.09.09.02 fixed this.

I'm closing this.

closed 2018-05-01 by cxbrooks