realsenseai/librealsense #1594: have no java wrapper

xiaosongfu, 2018-04-23:
i want to use java program language, there is no java wrapper ？

dorodnic, 2018-04-23:
Hi @xiaosongfu 
At the moment there isn't one (at least not an official one). We might add it in later releases, based on community feedback and demand.

xiaosongfu, 2018-04-23:
Thank you for your reply ^_^

Gnimuc, 2018-04-23:
It looks like @edwinRNDR is working on this: https://github.com/edwinRNDR/librealsense

cansik, 2018-07-16:
Compiling works also on MacOSX if you change the gradle script and add a new target for it.

```
...
$.platforms.all {
    jniLibSpec.targetPlatform it.name
}

binaries.all {
    lib new JNIDependency(project)
    cppCompiler.args "-I${project.projectDir}/../../include"
    linker.args "-L${project.projectDir}/../../"
    //linker.args "-lrealsense2"

    if (targetPlatform.operatingSystem.macOsX) {
    	cppCompiler.args "-I${project.projectDir}/../../include"
        linker.args "${project.projectDir}/../../build/Debug/librealsense2.dylib"
    }

    if (targetPlatform.operatingSystem.windows) {
...
```

I am going to give it a try :)

edwinRNDR, 2018-07-17:
@cansik Thanks. I have focussed on Windows support only, the C API (which is used by my Java wrapper) on macOS has had some issues that have recently been addressed (see: https://github.com/IntelRealSense/librealsense/issues/1586). 

Let me know if you can get it to work. Proper support for macOS and linux-x64 (with portable native libraries distributed in jar files) is on my wish list but I haven't had the time to figure out the build process for those.

cansik, 2018-07-17:
@edwinRNDR Compiling works and running too, after removing the absolute `System.load` commands and add relative ones (which target to the mac binaries too). I could create a pull request to your repository with the fixes for mac osx.

I also had the bug #1586 but could workaround it with a sleep in kotlin. My idea is to write a wrapper for [processing](https://processing.org/), so it would be nice to have a final jar with all the prebuilt binaries.

cansik, 2018-07-17:
@edwinRNDR I've added all my changes to the gradle build, readme and a multi-os-native library code snippet to the `Native` class.

The problem I am currently facing is, that the `librealsense2.dylib` is not automatically bundled when building everything with gradle. I could also not directly find the code in the build script where it will be bundled? Because of that I can not create the PR, because I can not test the multi-os-native-lib-loading. Could you tell me where you bundle all the dependencies?

I also would recommend that we change the name of the jar file to more descriptive name, for example: `librealsense2java.jar`

Here is the repository with the changes: https://github.com/cansik/librealsense/tree/master/wrappers/java

edwinRNDR, 2018-07-17:
@cansik Thank you. Bundling the native libraries is currently done by hand, exactly in the way as you describe it. This is less than ideal, but I should be able to add that step to the gradle build script. 

Is it OK if I merge from your repository instead of a pull request?

The name of the jar can be changed. I think it would also be easier to create a separate jar for each platform instead of a monolithic one as it is really hard to get Windows/macOS/Linux binaries in one place in a single build.

cansik, 2018-07-18:
@edwinRNDR What is in your eyes the advantage of merging by hand over applying the PR? If you want to, I could remove the auto-formatting of the `Native.java` file.

Ok, so I will check if I can add the possibility of bundling to the build script.

edwinRNDR, 2018-07-18:
@cansik Discard that :) I didn't see your pull request at the time I wrote that.

cansik, 2018-07-18:
@edwinRNDR Maybe it would make sense to open the issue page on your repository, so we can discuss the next features to implement, instead always writing here. I would like to contribute more to the library, and already implemented some methods on my branch.

RealSense-Customer-Engineering, 2018-10-09:
[Realsense Customer Engineering Team Comment]
@Xiaosong Sorry, the official release plan of java wrapper is still not clear yet. If you want to keep this open to wait for official release, we can keep it open.

edwinRNDR, 2018-10-12:
Hi @dorodnic linking it is good. But a word of caution about the state of the project as I haven't been able to dedicate my time to it: The project has partial support for a likely older version of the C API and only Windows binaries are provided. It however was good enough for me to use in production; YMMV

RealSense-Customer-Engineering, 2018-10-24:
[Realsense Customer Engineering Team Comment]
@Xiaosong As Dorodnic replied, we don't have the plan to add java wrapper in the near future. Considering this, is it Ok  for you to close the ticket at present?  You might check the release note if the feature is available later. Thanks!

dorodnic, 2018-10-24:
[wrappers#community-projects](https://github.com/IntelRealSense/librealsense/tree/development/wrappers#community-projects)

kasperkamperman, 2018-10-24:
@cansik I've found a Processing example for Windows (https://github.com/chungbwc/Magicandlove/tree/master/ml20181024b). Are you working on Mac? It would be a great to have the RealSense camera's working in Processing.

cansik, 2018-10-25:
@kasperkamperman Yes, this windows example uses the prebuilt library of @edwinRNDR. I already implemented a processing version of it, you can find it here, also with the [prebuilt java library for mac](https://github.com/cansik/realsense-processing/tree/master/lib) of the wrapper (the example code is in kotlin): [realsense-processing](https://github.com/cansik/realsense-processing)

And I already used it in a real world project ([RealSenseCamera.kt](https://github.com/bildspur/led-forest3/blob/master/src/main/kotlin/ch/bildspur/ledforest/realsense/io/RealSenseCamera.kt)) and it worked without any problems.

I could implement a processing library in the future, but I am not sure if it's already necessary, because we can just grab the image streams, without post processing. The java wrapper is not that far yet.

For the processing library: It would be necessary to process the streams in a second thread, because we have to copy the pixels one by one to the `PImage`, which is really slow.

RealSense-Customer-Engineering, 2018-11-16:
[Realsense Customer Engineering Team Comment]
@Xiaosongfu Is it ok for you to close this ticket?

closed 2018-11-27 by RealSense-Customer-Engineering