realsenseai/librealsense #1597: Not able to start recording

rafaor29, 2018-04-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Operating System & Version |  Linux (Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.13.0-38-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  {  2.10.4 }                          | 

### Issue Description
Whenever i try to start recording the following message appears in the screen:
Unknown in rs2_create_record_device(device:0x7f27cc0008c0, file:/):Error opening file: /root/Documents/20180423_153852.bag

How can i solve it?

dorodnic, 2018-04-23:
This might be a permission issue. Just as a quick check, you could retry with `sudo` and see if it works.

JasonChang13, 2018-04-30:
hi @rafaor29,

is running with sudo works for you?
thanks.

Regards,
Jason
Intel RSG Customer Platform Engineering