realsenseai/librealsense #1599: "feature" request - can you please upgrade to easylogger++ v9.96.4

omerbrandis, 2018-04-24:
Dear librealsense support,

i've been having difficulty integrating software that uses the latest version of easylogger (9.96.4) with librealsense.
i think it has something to do with the fact the librealsense is using a one year old version of easylogger, 
but more important that the "age" is the importance of the bug fixes made to it since then.

( for more info about the actual error i'm dealing with, you are welcome to view https://github.com/muflihun/easyloggingpp/issues/639)

but it seems to me that using easylogger 9.96.4 is a good idea regardless of my problems.
:-)

thanks,
omer.

dorodnic, 2018-04-24:
Hi @omerbrandis 
That makes sense. We will try to add this in one of the upcoming releases. You can disable librealsense ELPP via CMake flag if you are not using it and it is causing you problems for now.

omerbrandis, 2018-04-25:
hello MR.Dorodnic,

can you please provide more information about disabling usage of easylogger?
( i've seen parm BUILD_EASYLOGGINGPP for the build of librealsense itself, 
but i'd rather avoid building librealsense from source, so far i've used the package installation ).

thanks,
Omer.

dorodnic, 2018-04-25:
Yes, I was referring to `BUILD_EASYLOGGINGPP` Cmake flag

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@omerbrandis
do you still need any assist for this issue?

omerbrandis, 2018-05-15:
Hi,

No , i do not, 
do you wish me to close this issue?
( i thought it should be kept open until its executed)

omer.

RealSense-Customer-Engineering, 2018-05-16:
[Realsense Customer Engineering Team Comment]
@omerbrandis
This is new feature request, we will discuss internally, but might not have it in short time. I will create feature request internally, you can trace the release note of the new release in the future. So the ticket should be able to close.