realsenseai/librealsense #1600: [D415] Sync issue after updating to SDK 2.10.X

zuru, 2018-04-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.08.05.15 & 05.09.02.00 | 
| Operating System & Version |   Win (8.1/10)  |                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.4                           | 

### Issue Description
Hello,

We are having an issue after updating the SDK from version 2.9.0 to any variant of 2.10.**X**. Our issues are related to the updates in the syncer (namely the changes in _sync.h_ and _sync.cpp_ and the introduction of the _matcher_ system). More specifically, since we are integrating the SDK into our own computer vision framework, we have grouped its components (e.g. devices, sensors) into our own decomposition which also uses some static variables for the context and these components. When using either a _rs2::asynchronous_syncer_ or _rs2::syncer_ in our framework we get a stack overflow exception. However, in our initial implementation using SDK version 2.9.0, this was not the case. The error only manifests when updating to SDK version 2.10.**X** (i.e. with the introduction of the sync updates). It also manifests in both Windows 8.1 and Windows 10, with or without the Win10 SDK for the metadata extraction.

What is confounding is that when opening a sensor using a simple example, everything works OK even with SDK version 2.10.**X**. We believe we have tracked the issue in the _composite_matcher_ class and namely within the _find_matcher_ function. In the erroneous case, the _matcher_ does not exist in the _map_ using that stream's ID and it somehow enters an end-less callback & _composite_matcher_ creation loop that ends up in the stack overflow exception. 

I am attaching a code sample reproducing this issue using SDK version 2.10.**X**. There are 2 functions, one working okay, and another one reproducing the error that is resembling our framework design. The only dependency is the header only _cpplinq_ library.

**reproduce_sync_error.h**:
```
#pragma once

#include <librealsense2\rs.hpp>
#include <librealsense2\rs_advanced_mode.hpp>

#include <cpplinq.hpp>

#include <memory>
#include <iostream>

namespace rs2namespace {

	using rs2_profile_vector = std::vector<rs2::stream_profile>;

	class device_manager {
		static std::unique_ptr<rs2::context> rs2_context;
		static rs2::device_list rs2_devices;
		static std::vector<rs2::sensor> rs2_sensors;
		static std::map<rs2_stream, rs2_profile_vector> rs2_profiles;
		static std::vector<rs2::sensor> rs2_color_sensors;
		static std::vector<rs2::sensor> rs2_depth_sensors;
		static rs2::sensor selected_color_sensor;
		static rs2::sensor selected_depth_sensor;
		static rs2::device selected_device;
	public:
		static void start_erroneous_sync()
		{
			if (!rs2_context)
			{
				rs2_context = std::make_unique<rs2::context>();
				rs2_devices = rs2_context->query_devices();
				rs2_sensors = rs2_context->query_all_sensors();
				rs2_color_sensors = cpplinq::from(rs2_sensors)
					>> cpplinq::where([](rs2::sensor s) -> bool {
					return s.get_stream_profiles().back().stream_type()
						== rs2_stream::RS2_STREAM_COLOR;
				})
					>> cpplinq::to_vector();
				rs2_depth_sensors = cpplinq::from(rs2_sensors)
					>> cpplinq::where([](rs2::sensor s) -> bool {
					return s.get_stream_profiles().back().stream_type()
						== rs2_stream::RS2_STREAM_DEPTH;
				})
					>> cpplinq::to_vector();
				rs2_profiles = cpplinq::from(rs2_sensors)
					>> cpplinq::select([](rs2::sensor s) { return s.get_stream_profiles(); })
					>> cpplinq::to_map([](rs2_profile_vector ps) { return ps.back().stream_type(); });
				selected_device = rs2_devices.back();
				auto sensors = selected_device.query_sensors();
				auto selected_color_sensor = cpplinq::from(sensors)
					>> cpplinq::first([](rs2::sensor s) -> bool {
					return s.get_stream_profiles().back().stream_type()
						== rs2_stream::RS2_STREAM_COLOR;
				});
				auto selected_depth_sensor = cpplinq::from(sensors)
					>> cpplinq::first([](rs2::sensor s) -> bool {
					return s.get_stream_profiles().back().stream_type()
						== rs2_stream::RS2_STREAM_DEPTH;
				});
				auto selected_depth_profile = cpplinq::from(rs2_profiles.at(rs2_stream::RS2_STREAM_DEPTH))
					>> cpplinq::first([&](rs2::stream_profile p) -> bool {
					return p.fps() == 30
						&& p.stream_type() == rs2_stream::RS2_STREAM_DEPTH
						&& p.format() == rs2_format::RS2_FORMAT_Z16
						&& p.as<rs2::video_stream_profile>().width() == 640
						&& p.as<rs2::video_stream_profile>().height() == 480;
				});
				auto selected_color_profile = cpplinq::from(rs2_profiles.at(rs2_stream::RS2_STREAM_COLOR))
					>> cpplinq::first([&](rs2::stream_profile p) -> bool {
					return p.fps() == 30
						&& p.stream_type() == rs2_stream::RS2_STREAM_COLOR
						&& p.format() == rs2_format::RS2_FORMAT_BGR8
						&& p.as<rs2::video_stream_profile>().width() == 640
						&& p.as<rs2::video_stream_profile>().height() == 480;
				});
				selected_color_sensor.open(selected_color_profile);
				selected_depth_sensor.open(selected_depth_profile);
				rs2::asynchronous_syncer sync;
				sync.start([](rs2::frameset fs) {
					std::cout << "frame no: " << fs.get_frame_number() << std::endl;
				});
				selected_color_sensor.start(sync);
				selected_depth_sensor.start(sync);
			}
		}

		static void start_working_sync()
		{
			rs2::context ctx;
			auto devices = ctx.query_devices();
			auto device = devices.back();
			rs2::asynchronous_syncer sync;
			sync.start([&](rs2::frame f) -> void
			{
				if (!f.is<rs2::frameset>()) { return; }
				auto frameset = f.as<rs2::frameset>();
				auto depth_frame = frameset.get_depth_frame();
				auto color_frame = frameset.get_color_frame();
				if (depth_frame && color_frame) {
					std::cout << "frame no: " << f.get_frame_number() << std::endl;
				}
			});
			for (auto sensor : device.query_sensors()) {
				sensor.set_option(rs2_option::RS2_OPTION_FRAMES_QUEUE_SIZE, 0);
				auto info = rs2_camera_info::RS2_CAMERA_INFO_NAME;
				auto name = std::string(sensor.get_info(info));
				auto isDepth = name.find("Stereo") != std::string::npos;
				auto profile = cpplinq::from(sensor.get_stream_profiles())
					>> cpplinq::first([&](rs2::stream_profile p) -> bool
				{
					return p.fps() == 30.0 && p.as<rs2::video_stream_profile>().width() == 640
						&& p.as<rs2::video_stream_profile>().height() == 480
						&& p.format() == (isDepth ? rs2_format::RS2_FORMAT_Z16 : rs2_format::RS2_FORMAT_BGR8)
						&& p.stream_type() == (isDepth ? rs2_stream::RS2_STREAM_DEPTH : rs2_stream::RS2_STREAM_COLOR);
				});
				sensor.open(profile);
				sensor.start(sync);
			}
			while (true) {
				std::this_thread::sleep_for(std::chrono::milliseconds(33));
			}
		}
	};

	std::unique_ptr<rs2::context> device_manager::rs2_context = nullptr;
	rs2::device_list device_manager::rs2_devices = rs2::device_list();
	std::vector<rs2::sensor> device_manager::rs2_sensors = std::vector<rs2::sensor>();
	std::map<rs2_stream, rs2_profile_vector> device_manager::rs2_profiles = std::map<rs2_stream, rs2_profile_vector>();
	std::vector<rs2::sensor> device_manager::rs2_color_sensors = std::vector<rs2::sensor>();
	std::vector<rs2::sensor> device_manager::rs2_depth_sensors = std::vector<rs2::sensor>();
	rs2::device device_manager::selected_device = rs2::device();
}
```
**main.cpp**:
```
#include <reproduce_sync_error.h>
int main(int argc, char** argv)
{
	rs2namespace::device_manager::start_erroneous_sync();
	return std::getchar();
}

int main2(int argc, char** argv)
{
	rs2namespace::device_manager::start_working_sync();
	return std::getchar();
}
```
Thanks for any help in advance,
Nick

dorodnic, 2018-04-25:
Hi Nick,
Thanks for the detailed feedback. @aangerma, please take a look ASAP.

aangerma, 2018-04-25:
Hi @zuru 
I investigated the issue and it can occur whenever the `stream_profile` object passed to `open` was created from a different instance of the same device. 

To illustrate, the following snippet will reproduce the same problem:
```cpp
rs2::context ctx;
rs2::syncer s;
rs2::device d1 = ctx.query_devices()[0];
rs2::device d2 = ctx.query_devices()[0];

d2.first<rs2::depth_sensor>().open(d1.first<rs2::depth_sensor>().get_stream_profiles().front());
d2.first<rs2::depth_sensor>().start(s);
s.wait_for_frames();
```
Despite `d1` and `d2` both being created from same physical device, they are treated by the `syncer` as distinct entities. As a result the `syncer` fails to handle frames with stream profile from a "different" device.

We should certainly resolve the crash, but as a work-around you should avoid referencing the same sensor from multiple API objects.

zuru, 2018-04-25:
Hi @aangerma,

The workaround seems to work. I also see that the bug has been fixed, and will logically be available in the next release.

Thank you,
Nick

closed 2018-04-25 by zuru