realsenseai/librealsense #1601: Obtaining RGB values fora colorised PCL point cloud.

conorjmcq, 2018-04-24:
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300} | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC|
| SDK Version                            |  { 2.10.3 }                          | 

### Issue Description
Hi there,

I'm using the SR300 for a college assignment and am stuck on a bit of an issue.

I am using PCL and have generated the required x, y, z coords for the pointcloud. However I would like to manipulate the data that I have captured depending on the individual RGB values at the desired point. I understand that get_texture_coordinates will return the U, V coords but is there a way to only extract the RGB values, or even obtain them  somehow ?

Thank you very much for you time.

dorodnic, 2018-04-25:
Hi @conorjmcq 
We don't have such an example out of the box, but the process should be very similar to [colorized PLY export](https://github.com/IntelRealSense/librealsense/blob/master/src/archive.cpp#L41)
More specifically please refer to [archive.cpp:57](https://github.com/IntelRealSense/librealsense/blob/master/src/archive.cpp#L57)

conorjmcq, 2018-04-25:
Hi @dorodnic ,

Thank you for the quick reply. I haven't managed to come to a solution yet unfortunately. You're link definitely helped me to understand the method, but I'm still miles off. 

https://github.com/IntelRealSense/librealsense/issues/1143

This link talks about the same issue but I don't think there's much mention of how they achieved it.

If there is anything else you could help me with then I'd greatly appreciate it. If not then thank you very much for your time.

Yonatan-Sade, 2018-05-08:
Hi. I am still striggling with it...
Can anyone help me with this? It would be very appreciated...
I want to convert data from the camera to XYZRGB point cloud of PCL.

I am starting with:
```
auto frames = pipe.wait_for_frames();
auto depth = frames.get_depth_frame();
points = pc.calculate(depth);
auto color = frames.get_color_frame();
pc.map_to(color);
auto vertices = points.get_vertices();      
auto tex_coords = points.get_texture_coordinates();
```
Then, using the `vertices `I can assign the XYZ values.
But How can I get the RGB?
I know that can be done by the `color `and `tex_coords`, but how **excactly**?
thank all for any support

Yonatan

conorjmcq, 2018-05-09:
> I think I got it!
> Try this code snippet. worked for me (not all the points got valid color -
> any idea?):
>
> #include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
> #include "../../../examples/example.hpp" // Include short list of convenience functions for rendering
> #include <pcl/point_types.h>
> #include <pcl/filters/passthrough.h>
> #include <pcl/visualization/pcl_visualizer.h>
>
> std::tuple<uint8_t, uint8_t, uint8_t> get_texcolor(rs2::video_frame texture, rs2::texture_coordinate texcoords)
> {
> 	const int w = texture.get_width(), h = texture.get_height();
> 	int x = std::min(std::max(int(texcoords.u*w + .5f), 0), w - 1);
> 	int y = std::min(std::max(int(texcoords.v*h + .5f), 0), h - 1);
> 	int idx = x * texture.get_bytes_per_pixel() + y * texture.get_stride_in_bytes();
> 	const auto texture_data = reinterpret_cast<const uint8_t*>(texture.get_data());
> 	return std::tuple<uint8_t, uint8_t, uint8_t>(
> 		texture_data[idx], texture_data[idx + 1], texture_data[idx + 2]);
> }
>
> pcl::PointCloud<pcl::PointXYZRGB>::Ptr points_to_pcl(const rs2::points& points, const rs2::video_frame& color)
> {
> 	pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZRGB>);
>
> [... 58 lines cut ...]
> 		boost::this_thread::sleep(boost::posix_time::microseconds(1));
> 	}
>
> 	return EXIT_SUCCESS;
> }

Hi there,

I gave up that pursuit and decided that I was going to color the points
based on depth. It turned out how I wanted it, but thank you very much for
sending this code chunk. I don't know how helpful I may be as I still have
very little experience. I'm currently trying to implement SLAM with the
Realsense.

I'll try your code snippet once I get into College.

Thanks again!
Conor.

GKachaner, 2018-05-09:
Hello ! 
I had the same issue and i solved it using this function : 

```
typedef pcl::PointXYZRGB PointT;
pcl::PointCloud<PointT>::Ptr points_to_pcl(const rs2::points& points)
{
	pcl_ptr cloud(new pcl::PointCloud<PointT>);

	auto sp = points.get_profile().as<rs2::video_stream_profile>();
	cloud->width = sp.width();
	cloud->height = sp.height();
	cloud->is_dense = false;
	cloud->points.resize(points.size());
	auto ptr = points.get_vertices();
	for (auto& p : cloud->points)
	{
		p.x = ptr->x;
		p.y = ptr->y;
		p.z = ptr->z;
		ptr++;
	}

	return cloud;
}
```
The trick is in the type of points you are using with PCL.
If you stick to PointXYZ you won't store the color information.
Using PointXYZRGB should do it.
Hope it works for you.

Grégoire

closed 2018-05-16 by RealSense-Customer-Engineering