realsenseai/librealsense #1607: Test

RobO-Intel, 2018-04-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RobO-Intel, 2018-04-25:
ad

RobO-Intel, 2018-04-25:
test

RobO-Intel, 2018-04-25:
test

RobO-Intel, 2018-04-25:
test

RobO-Intel, 2018-04-25:
[Realsense Customer Engineering Team Comment]

3

RobO-Intel, 2018-04-26:
test

RealSense-Customer-Engineering, 2018-04-26:
[Realsense Customer Engineering Team Comment]

reply

RobO-Intel, 2018-04-26:
conversation

RealSense-Customer-Engineering, 2018-04-27:
[Realsense Customer Engineering Team Comment]

Reply

RobO-Intel, 2018-04-27:
Label

RobO-Intel, 2018-04-27:
label Test

RobO-Intel, 2018-04-29:
.

RealSense-Customer-Engineering, 2018-04-30:
test

RealSense-Customer-Engineering, 2018-04-30:
test

RealSense-Customer-Engineering, 2018-04-30:
[Realsense Customer Engineering Team Comment]
p

closed 2018-05-03 by RobO-Intel