realsenseai/librealsense #1608: rs-capture compiling error

sallvadorrr, 2018-04-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |  Linux Ubuntu 16.04| 
| Kernel Version (Linux Only)    |  4.13.0-36-generic              | 
| Platform                                 | PC |
| SDK Version                            | 2.10.4                          | 

### Issue Description
../third-party/glfw is added to the resources in the project.
When compiling rs-capture.cpp in CodeBlocks I get errors

_No supported client library selected. 
No supported window creation API selected._ 

in /third-party/glfw/src/internal.h

Thank you for any suggestions.

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
@sallvadorrr  Could you please check if "third-party/glfw/src/glfw_config.h" exist in your project?

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
@sallvadorrr  Could you please feedback if "third-party/glfw/src/glfw_config.h" exist in your project?

sallvadorrr, 2018-06-07:
@RealSense-Customer-Engineering Sorry for a late response! At this point I am not sure what was wrong, probably I didn't link resources properly in CodeBlocks. It worked later alright as far as I can remember!

closed 2018-06-07 by sallvadorrr