realsenseai/librealsense #1610: D435 SDK2.0 with Intel stick computer

Loriskar, 2018-04-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D410/D435 } | 
| Firmware Version                   | (uptodate) | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (NA)                                         | 
| Platform                                 | PC/  |
| SDK Version                            |  uptodate }                          | 

### Issue Description
I am trying to build up a portable Realsense camera platform with Intel stick computer. I have an Intel Realsense D410 module and a D435 camera on the way. If it is just for image recording purpose, what is the requirement of the CPU? Would the Atom™ x5-Z8300 be sufficient? (I noticed a version of computer stick with core m and realsense on IDF16, so I can I assume that Core M should work fine with the camera? )
I was thinking about using ARM for the same purpose last year and had some supporting issue. So I hope someone could help me with this issue ahead of time.

agrunnet, 2018-04-25:
Yes. That should work provided it has usb3.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@Loriskar, I am closing this issue but if you still need more information feel free to open a new issue.

closed 2018-05-15 by RealSense-Customer-Engineering