realsenseai/librealsense #1612: rs-measure example not working

AbhijithNB, 2018-04-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.3  | 

### rs-measure example crashes
All the other examples are working fine, but only "rs-measure" is not working. It gives a bunch of  "Cannot find or open the PDB file." errors for various DLLs (C:\Windows\System32\ntdll.dll, kernel32.dll, glu32.dll, win32u.dll, user32.dll etc...). Then a bunch of "The thread 0x2774 has exited with code 0 (0x0)." errors for various Threads. And finally I get this error "The program '[13960] rs-measure.exe' has exited with code 1 (0x1)."

I used both 2.10.3 and 2.10.4 SDKs. I still get same errors.

Also, on a different note, is there any way that I can get all the pixels or PointClouds which are colored Blue and Cyan in the Colorized Depth frame?

adarshmuthappa, 2018-04-27:
After I updated the D415 firmware to 5.9.9.2, all examples work. Try that. Release notes state that firmware>5.9.9+ is required for 2.10.4 SDK version and >5.9.2+ for 2.10.3 SDK version.

https://github.com/IntelRealSense/librealsense/releases

dorodnic, 2018-04-29:
These might be an issue with this example in 2.10.4 (due to reported bug in hole-filling), but it should be resolved in the next update.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@AbhijithNBC
Have you tried the 2.11.0 release? it should fix the problem.

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
@AbhijithNBC
Any update with the test of latest LibRS?

AbhijithNB, 2018-05-24:
Its working after I updated the firmware.
Thank you for the support.

closed 2018-05-24 by AbhijithNB