realsenseai/librealsense #1613: using realsense2.dll

deusxmachina42, 2018-04-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.10                           | 

### Issue Description

I try to compile the "Hello World" program for .NET Wrapper, but it is not possible to referecne realsense2.dll. With Intel.RealSense.dll there is no problem, but whenever i try to reference realsense2.dll, the compiler throws this error message (translated)

It was not possible to add a reference to C:\...\realsense2.dll . Make sure, you can access the file and it is an valid assembly or COM-component

I tried with both, the downloaded .dll of the SDK and the self-compiled one.

Thank you!

dorodnic, 2018-04-29:
Hi @deusxmachina42 
Due to how .NET interoperability works you need both Intel.RealSense.dll and realsense2.dll at run-time. 
However, only the Intel.RealSense.dll is a .NET assembly you can (and should) add reference to. 
realsense2.dll is native (C++) library that must be found at runtime (at current directory usually). You can't directly reference it in your C# project. 
Please note that you also must select architecture (x86 or x64) that matches how realsense2.dll was compiled. Using Any CPU is not recommended when embedding native components.

dorodnic, 2018-05-06:
Hi @deusxmachina42 
Do you need further assistance with this issue?

closed 2018-05-08 by dorodnic

Georgidg, 2018-08-23:
Hi @dorodnic 
I have the same issue that @deusxmachina42  has.
when i try to run the sample code that download from github, the program throws runtime  exep
[image]
tion that the realsense2.dll are not found(you said up that it's not possible to directly reference it in  C# project).
I made sure that my program and "realsense2.dll" are both x64.
can you have any idea whats the problam?

snmathur4, 2019-03-13:
Hi, where are we supposed to put the realsense2.dll exactly? Just in the debug/release folder manually?

voqk, 2019-09-23:
I had this problem with the csharp/cs-tutorial-1-depth project. I copied realsense2.dll and realsense2.pdb from the installed SDK folder: `C:\Program` Files (x86)\Intel RealSense SDK 2.0\bin\x64` into the project folder: `librealsense\build\wrappers\csharp\cs-tutorial-1-depth`. Then in Visual Studio I right clicked the cs-tutorial-1-depth project, Add -> Existing Item, and finally set the "Copy to Output Directory" property of both files to "Copy if newer".

Joey-Huang713, 2019-09-24:
hi @voqk you can try to copy realsense2.dll into the folder:\librealsense-master\build\Debug.
it just works for me , good luck.

voqk, 2019-09-25:
hi @Joey-Huang713, sorry for the confusion. I did have this problem, but solved it with the steps in my comment. Just wanted to share the love. (I'll edit for clarity)

This is a great little sensor and the .NET API wrapper is a big plus!