realsenseai/librealsense #1620: c#-wrapper - Frame did not arrive in time!

Tkong1, 2018-04-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                         | 

### Issue Description
While debugging the *cs-tutorial-1-depth*-tutorial from the c#-wrapper an exception appears after some seconds. See:

System.Exception
  HResult=0x80131500
  Message=Frame did not arrive in time!
rs2_wait_for_frame(queue:000000001D8C0520)
  Source=<Cannot evaluate the exception source>
  StackTrace:
<Cannot evaluate the exception stack trace>

RealSense-Customer-Engineering, 2018-04-30:
[Realsense Customer Engineering Team Comment]
hi @Tkong1,

could you please tell if RealSense Viewer also have this issue on your PC or only the C#-wrapper sample?
thanks.

Tkong1, 2018-05-03:
hi @RealSense-Customer-Engineering ,
in RealSense Viewer after some seconds also an exception appears:

Oops, something went wrong!
RealSense error calling:
Unknown in rs2_close(sensor:000002568359BD60): profile not found

But the Error appears not always.

RealSense-Customer-Engineering, 2018-05-04:
[Realsense Customer Engineering Team Comment]
hi @Tkong1,
since Realsense Viewer have issue, I think we should fix this before we move to C# wrapper issue..
would like to confirm before realsense viewer report error, do you see the D435 depth stream?
please also confirm your cable is USB3 certified.
thanks.

RealSense-Customer-Engineering, 2018-05-21:
[Realsense Customer Engineering Team Comment]
hi @Tkong1,

do you still have this issue with latest release now?

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any follow question.

closed 2018-06-11 by RealSense-Customer-Engineering