realsenseai/librealsense #1624: auto_exposure problem

kuonangzhe, 2018-04-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Platform                                 | PC |
| SDK Version                            |  2.0                      | 

I am using the python wrapper for a project, where I want to set the enable_auto_exposure first so that the camera can adjust to the proper exposure value, then disable the setting and confirm to use the adjusted exposure. However, it does not work. It seems like the auto_exposure only adjust something else but actually not tuning the exposure value. 

code snippets are as follows:
```
print(sensor.get_option(rs.option.exposure)
sensor.set_option(rs.option.enable_auto_exposure, 1)
# Then take some images to tune the exposure
# ...
# Then disable the setting
sensor.set_option(rs.option.enable_auto_exposure, 0)
print(sensor.get_option(rs.option.exposure)
```
 I found that the printed exposure value did not change. I also tried this in realsense-viewer, with the same idea, and save to the config.json. The value in .json still not change as above. Are there any suggestions?

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@kuonangzhe
I tested with the same result as you mentioned, i will check internally if there is any way to get back the changed exposure when auto ep is enabled/disabled.

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
@kuonagzhe
Does the response answer your question?

closed 2018-06-05 by RealSense-Customer-Engineering

henryjchang, 2018-06-05:
I tried the code snippet to get exposure value from metadata and am not getting out varying exposure values.

On Ubuntu 16.04, running librealsense v2.10.0, with firmware v5.8.14 on a D435,
I just get a constant depth exposure of 8500.

On Ubuntu 16.04, running librealsense v2.10.0, with firmware v5.9.11 on a D410,
I just get a constant depth exposure of 33000.

The output after
```
dev.query_sensors()[0].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
 std::this_thread::sleep_for(std::chrono::seconds(1));
 std::cout << "exposure = " << dev.query_sensors()[0].get_option(rs2_option::RS2_OPTION_EXPOSURE) << std::endl;
```
is just the same respective exposure value for each camera, regardless of setting the option to `1` or `0`.

Any idea what could be going wrong?