realsenseai/librealsense #1626: How to correctly use C API for processing block?

Gnimuc, 2018-04-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |    MacOS  | 
| Kernel Version (Linux Only)    |  NA                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.2                        | 

### Issue Description
I'm trying to colorize the depth stream via colorizer processing block using C API. Here is an MWE:
```C
rs2_processing_block* colormap = rs2_create_colorizer(&e);
check_error(e);
rs2_frame_queue* frame_queue = rs2_create_frame_queue(1, &e);
check_error(e);

rs2_frame* frames = rs2_pipeline_wait_for_frames(pipeline, 5000, &e);
check_error(e);
int num_of_frames = rs2_embedded_frames_count(frames, &e);
check_error(e);

int i;
for (i = 0; i < num_of_frames; ++i)
{
    // The retunred object should be released with rs2_release_frame(...)
    rs2_frame* frame = rs2_extract_frame(frames, i, &e);
    check_error(e);

    // Check if the given frame can be extended to depth frame interface
    // Accept only depth frames and skip other frames
    if (0 == rs2_is_frame_extendable_to(frame, RS2_EXTENSION_DEPTH_FRAME, &e))
         continue;

    rs2_start_processing(colormap, (rs2_frame_callback*)frame_queue, &e);  
    check_error(e);
    rs2_frame_add_ref(frame, &e);
    check_error(e);
    rs2_process_frame(colormap, frame, &e); 
```
The code crashed with the error: **Thread 1: EXC_BAD_ACCESS (code=1, address=0x0)**
[image]

Did I use the API correctly?  I have no idea on how to debug this. Any help or hint would be appreciated!

Gnimuc, 2018-04-30:
@dorodnic thanks for your quick reply!!! I changed `rs2_start_processing` to `rs2_start_processing_queue` and everything works like a charm.

dorodnic, 2018-05-30:
Hi @Gnimuc 
The flow seems to be correct, however I'm missing calls to `rs2_release_frame` or `rs2_keep_frame`. These must be called on every frame in the frameset (even ones you choose to ignore) and on the frameset itself. 
Without it, you are likely to get stuck after 16 frames (this is the limit of how many frames you are allowed to use at once without releasing or keeping).
Let me know if it helps, sorry for some delay

Gnimuc, 2018-05-30:
yes, you're right. Adding the `rs2_release_frame` fixes the error. Thanks for your explanation!

closed 2018-05-30 by Gnimuc

meimar18, 2018-11-23:
hi there! 

Related to this question I was wondering how to set the callbacks for receiving the processed images there instead of polling the framequeue as you pointed in the first question. Whould it work with something like:

void myCallback(rs2_frame* processedFrame){...} 

rs2_start_processing( alignProcessingBlock, (rs2_frame_callback*) myCallback, &e);

and just calling rs2_process_frame(alignProcessingBlock, frame, &e); whould call my callback when finished processing?

Thanks !