realsenseai/librealsense #1630: Patch timeout error

arunabhcode, 2018-04-30:
Command : ./scripts/patch-realsense-ubuntu-xenial.sh

Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
git is already the newest version (1:2.7.4-0ubuntu1.3).
linux-headers-generic is already the newest version (4.4.0.122.128).
The following packages were automatically installed and are no longer required:
  libappindicator1 libglfw2 libindicator7 linux-headers-4.13.0-37 linux-headers-4.13.0-37-generic linux-headers-4.4.0-121 linux-headers-4.4.0-121-generic linux-image-4.13.0-37-generic
  linux-image-4.4.0-121-generic linux-image-extra-4.13.0-37-generic linux-image-extra-4.4.0-121-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Cloning into './ubuntu-xenial-hwe'...
fatal: unable to connect to kernel.ubuntu.com:
kernel.ubuntu.com[0: 91.189.94.216]: errno=Connection timed out

akatrevorjay, 2018-04-30:
Do you see what the issue is? ;) It's merely trying to clone the official ubuntu kernel git, and for some reason you're getting a timeout doing so (It's online right now, so this is likely on your side).
I would just try it again, it's probable it will download this time.

arunabhcode, 2018-05-01:
There was a firewall issue.

closed 2018-05-01 by arunabhcode

padhupradheep, 2020-09-07:
Im facing this issue in ubuntu 16.04