realsenseai/librealsense #1631: Opcodes do not match - Stereo No Longer Working

juanBerger, 2018-05-01:
--------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |{D415}| 
| Firmware Version                   |{05.09.09.02}| 
| Operating System & Version | {Ubuntu 16.04}| 
| Kernel Version (Linux Only)    | {4.13.0-39-generic} | 
| Platform                                 | {upCore}  |
| SDK Version                            |  { 2.10.4}                          | 

### Issue Description
Hi All - 
After trying a longer cable I get the following message in the rs viewer:

Invalid Value in rs2_get_option_range(options:000002B3641140C0, option:Stereo Baseline, min:0.0001, max:0.01, step:1e-06, def:0.001):
OpCodes do not match! Sent 21 but received -6!

Went back to the original cable (that came with the D415) and still have the same problem. I reinstalled the firmware, same issue; and tried the calibration tool but that fails with "calibration tables on device are not supported". The RGB camera works fine but the stereo looks like it does here #1508. Any way to solve this? Thx 

p.s Also tried this on other hardware w/ Ubuntu 16.04, same issue.

RealSense-Customer-Engineering, 2018-05-07:
[Realsense Customer Engineering Team Comment]
hi @juanBerger,

I think we need to make sure your camera back to normal for the calibration first.
can you please try the calibration tool,https://downloadcenter.intel.com/download/27415/?v=t
with Intel.Realsense.CustomRW.exe -g to recover the calibration table and see if it recovered.
and try calibration again to see if you can calibrate the camera now.

juanBerger, 2018-05-10:
Hi, 
Thank you for the response, this appears to have worked. After running 
`Intel.Realsense.CustomRW.exe -g`
I was able to do the calibration. Using the depth quality tool I'm getting an RMS error of about 40mm - is this typical? (measuring a distance of ~3m). Thanks

RealSense-Customer-Engineering, 2018-05-10:
[Realsense Customer Engineering Team Comment]
hi @juanBerger,

great, 40mm error at 3m is a very good result, so back to your main question, is the issue still there? with long or short cable?
thanks.

juanBerger, 2018-05-12:
Hi - I no longer see the error and both cables work mostly without issue. Frequently though, the camera will appear as a 2.1 device. Simply reversing the orientation of the c side of the plug has been fixing the problem, though. This happens with both cables and on multiple host machines. Thx 
>